e2nIEE/pandapower #1991: enhancements of protection module

pawellytaev, 2023-04-19:
### Issue Report Checklist

* [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

This issue is a general issue for code enhancements of pandapower.protection which are related to the current state of the module, i.e. not including new features. I see the following points as part of future improvement:

- [ ] removal of unnecessary deepcopy's in the code
- [ ] inclusion of relays as part of the net object (currently it is a list of dictionaries)
- [ ] acceleration of code by replacing loops with vectorized calculations

feel free to add further points related to the general protection module

vogt31337, 2024-11-21:
@pawellytaev @gbanerjeefraunhofer could you take a look at this and maybe just close it?

gbanerjeefraunhofer, 2024-12-16:
This issue is deprecated and already taken care of.

closed 2024-12-16 by gbanerjeefraunhofer