e2nIEE/pandapower #1992: [bug] Deprecated numpy attribute in CIM Converter

wenglerjoe, 2023-04-19:
## Problem Description
When trying to open a CIM file using the from_cim.from_cim function in the pandapower converter, I receive an AttributeError that module "numpy" has no attribute 'typeDict'
...\pandapower\converter\cim\cim_tools.py", line 38, in extend_pp_net_cim
    np_str_type = np.typeDict.get('str')

Apparently, typeDict has been deprecated and replaced with scttypeDict.

### What steps reproduce the problem?

1. Fresh install of python environment
2. import pandapower and from_cim function from pandapower.converter
3. Open CIM file (.zip file) using the from_cim.from_cim function 

### What is the expected output? What do you see instead?
Successfully opens CIM file and stores it in a pandapower net object.

### Paste Traceback/Error Below (if applicable)

```python-traceback

Traceback (most recent call last):
  File "H:\Dokumente\Python Scripts\PandaPower\test\example_cim\test.py", line 13, in <module>
    net = from_cim.from_cim(file_list=["202200707_CIM_CGMES_Testgebiet_NE7.zip"])
  File "H:\Dokumente\Python Scripts\PandaPower\lib\site-packages\pandapower\converter\cim\cim2pp\from_cim.py", line 76, in from_cim
    pp_net = from_cim_dict(cim_parser, convert_line_to_switch=convert_line_to_switch,
  File "H:\Dokumente\Python Scripts\PandaPower\lib\site-packages\pandapower\converter\cim\cim2pp\from_cim.py", line 29, in from_cim_dict
    return cim_converter.convert_to_pp(convert_line_to_switch=convert_line_to_switch, line_r_limit=line_r_limit,
  File "H:\Dokumente\Python Scripts\PandaPower\lib\site-packages\pandapower\converter\cim\cim2pp\build_pp_net.py", line 1886, in convert_to_pp
    self.net = cim_tools.extend_pp_net_cim(self.net, override=False)
  File "H:\Dokumente\Python Scripts\PandaPower\lib\site-packages\pandapower\converter\cim\cim_tools.py", line 38, in extend_pp_net_cim
    np_str_type = np.typeDict.get('str')
  File "H:\Dokumente\Python Scripts\PandaPower\lib\site-packages\numpy\__init__.py", line 320, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'. Did you mean: 'sctypeDict'?

```

## Versions

* pandas version: 2.0.0
* networkx version: 3.1
* scipy version: 1.10.1
* numpy version: 1.24.2
* packaging version: 23.1
* tqdm version: 4.65.0
* deepdiff version: 6.3.0
* Operating System name/version:  Windows 10

## Labels (if access available)
`bug`, `converter`
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

wenglerjoe, 2023-04-19:
Above was done in Python 3.10

Solution (for now): 
Python 3.9: installing numpy==1.21 which still supports the above function.

jwiemer112, 2023-05-13:
@wenglerjoe The bug is fixed now, have a look at PR #2005

Greetz Jan

closed 2023-05-13 by jwiemer112