e2nIEE/pandapower #1996: [bug] Issue when running DC powerflow after AC powerflow

BDonnot, 2023-04-19:
### Issue Report Checklist

* [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [X] Tried basic troubleshooting (if a bug/error)
    * [X] Restarted interpreter
    * [X] Checked the pythonpath
* [X] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description

When I run a DC powerflow after an AC powerflow, some columns of the `net.res_bus` (such as `vm_pu` and `q_mvar`) does not seem to be "updated" with the powerflow results.

This results in strange behaviour such as (see reproducible code below) `net.res_line["vm_from_pu"]` which is not equal to `net.res_bus.iloc[THE BUS WHERE THE LINE IS CONNECTED]["vm_pu"]`.

The doc (see *eg* https://pandapower.readthedocs.io/en/latest/elements/line.html#result-parameters) clearly specify : `vm_from_pu: voltage magnitude at from bus` which is not the case in the example.

### What steps reproduce the problem?

1.  install pandapower
2. run the code

### What is the expected output? What do you see instead?

If you look at the bus to which a line is connected and the value of "vm_from_pu" then there should not be a difference. The voltage magnitude is unique at a given bus.

```python
import pandapower as pp
import pandapower.networks as pn
import numpy as np

def get_installed_versions():
    res = []
    import pandapower as pp
    res.append(f"* pandapower version: {pp.__version__}")
    import pandas as pd
    res.append(f"* pandas version: {pd.__version__}")
    import networkx
    res.append(f"* networkx version: {networkx.__version__}")
    import scipy
    res.append(f"* scipy version: {scipy.__version__}")
    import numpy as np
    res.append(f"* numpy version: {np.__version__}")
    import packaging
    res.append(f"* packaging version: {packaging.__version__}")
    import tqdm
    res.append(f"* tqdm version: {tqdm.__version__}")
    import deepdiff
    res.append(f"* deepdiff version: {deepdiff.__version__}")
    return "\n".join(res)
print(get_installed_versions())

# now the test
net = pn.case14()

line_id = 6
# run an ac powerflow (simple setting)
pp.runpp(net, init="flat", lightsim2grid=False, distributed_slack=False)
# normal behaviour (see below for detail about this line of code)
from_bus, to_bus = net.line.iloc[line_id][["from_bus", "to_bus"]]
assert np.allclose(net.res_bus.iloc[[from_bus, to_bus]]["vm_pu"].values,
                   net.res_line.iloc[line_id][["vm_from_pu", "vm_to_pu"]].values)

# then run a dc powerflow
pp.rundcpp(net)
assert( net.res_line.iloc[line_id][["vm_from_pu", "vm_to_pu"]] == [1.0, 1.0]).all()
net.line.iloc[line_id][["from_bus", "to_bus"]]  # [3, 4]
from_bus, to_bus = net.line.iloc[line_id][["from_bus", "to_bus"]]  # [3, 4]
assert from_bus == 3
assert to_bus == 4
assert np.allclose(net.res_bus.iloc[[3, 4]]["vm_pu"], [1.017671, 1.019514])

# weird behaviour (same as above, in one line of code)
assert np.allclose(net.res_bus.iloc[[from_bus, to_bus]]["vm_pu"].values,
                   net.res_line.iloc[line_id][["vm_from_pu", "vm_to_pu"]].values)

```

### Paste Traceback/Error Below (if applicable)

```python-traceback
Traceback (most recent call last):
  File "test_pp_dc.py", line 48, in <module>
    assert np.allclose(net.res_bus.iloc[[from_bus, to_bus]]["vm_pu"].values,
AssertionError
```

## Versions

* pandapower version: 2.12.1
* pandas version: 1.5.3
* networkx version: 2.5
* scipy version: 1.6.0
* numpy version: 1.20.3
* packaging version: 20.9
* tqdm version: 4.61.2
* deepdiff version: 6.2.1
* Operating System name/version: ubuntu 20.04

ps. it whould be nice to have a helper function from pandapower from which we could copy paste results above. Such as:
```python
import pandapower as pp
print(pp.get_installed_versions())
```

And somewhere in pandapower you define:

```python
def get_installed_versions():
    res = []
    import pandapower as pp
    res.append(f"* pandapower version: {pp.__version__}")
    import pandas as pd
    res.append(f"* pandas version: {pd.__version__}")
    import networkx
    res.append(f"* networkx version: {networkx.__version__}")
    import scipy
    res.append(f"* scipy version: {scipy.__version__}")
    import numpy as np
    res.append(f"* numpy version: {np.__version__}")
    import packaging
    res.append(f"* packaging version: {packaging.__version__}")
    import tqdm
    res.append(f"* tqdm version: {tqdm.__version__}")
    import deepdiff
    res.append(f"* deepdiff version: {deepdiff.__version__}")
    return "\n".join(res)
```

## Labels (if access available)
* [X] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

rbolgaryn, 2023-05-10:
https://stackoverflow.com/questions/29751572/how-to-find-a-python-packages-dependencies

rbolgaryn, 2023-05-10:
pkg_resources is deprecated, importlib should be used instead. 
Here documentation to importlib, hopefully it can be used to solve this easily. If not, then it is worth to write own implementation for covenience.

https://docs.python.org/3/library/importlib.resources.html

https://docs.python.org/3/library/importlib.metadata.html

BDonnot, 2023-05-10:
Hello,

Maybe I have misunderstood your comment, but my issue was related to DC powerflow and not packages version or anything else that could be addressed by importlib or pkg_resources.

rbolgaryn, 2023-05-10:
```python

net._ppc["bus"][:, VM]
Out[78]: array([1., 1., 1., 1., 1., 1., 1., 1., 1.])

```

rbolgaryn, 2023-05-10:
This solves the issue:

```python
net = pp.networks.case9()
pp.runpp(net)
net.res_bus.drop(net.res_bus.index, inplace=True)
pp.rundcpp(net)
```

So the old value is coming from the net.res_bus table

rbolgaryn, 2023-05-10:
Making a change in _get_bus_results to always write results for vm_pu as following should solve it:

```python
def _get_bus_v_results(net, ppc, suffix=None):
    ac = net["_options"]["ac"]
    bus_idx = _get_bus_idx(net)

    res_table = "res_bus" if suffix is None else "res_bus%s" % suffix
    # if ac:
    #     net[res_table]["vm_pu"] = ppc["bus"][bus_idx][:, VM]
    net[res_table]["vm_pu"] = ppc["bus"][bus_idx][:, VM]
    # voltage angles
    net[res_table]["va_degree"] = ppc["bus"][bus_idx][:, VA]
```

This will result in having vm_pu = 1 in net.res_bus. 

Previously, the result was always np.nan (without the AC power flow). 
The results for vm_xxx_pu has always been taken from ppc["bus"][:, VM_PU] so it has always been 1.0

Does this seem right as a fix?

BDonnot, 2023-05-10:
Hello again 😊

My suggestion about having something to output the versions of the packages was because I needed this information to write correct issues. So this seemed fair to have an easy solution directly in pandapower. Hence the "quick and dirty" proposition I made.

For the DC powerflow your fix seems reasonable indeed. I suspected it was because some values were re used from the ax powerflow.

Thanks for the reactivity

PR #2017 (2023-05-12): fix bug with old values from AC power flow for results of DC power fl…
…ow for res_bus (fixes #1996)
commit 20251b65 (2023-05-12): fix bug with old values from AC power flow for results of DC power flow for res_bus (fixes #1996) (#2017)
closed 2023-05-12 by jwiemer112