e2nIEE/pandapower #1997: Unexpected plot result when modelling a chp as sgen to reduce the loading of transformer

Joboreh97, 2023-04-21:
Hello everyone. When I model an sgen as a generator to reduce the overloading of the transformer in my network,without the sgen ,I get a good result for the transformer overloading,but with the sgen,the transformer overloading is so large. My goal is for the sgen to be another current source,but from the look of things,it's overloading the transformer.

#Code for sgen
pp.create_sgen(net, bus0, p_mw=2, q_mvar=-0.5, name="static generator")

#Code for transformer
trafo1 = pp.create_transformer(net, bus1, bus2, name="10kV/0.4kV transformer", std_type="0.63 MVA 10/0.4 kV")

I would appreciate if someone can help me solve this problem. Thanks a lot

jwiemer112, 2023-04-24:
Hey @Joboreh97 ,

the chosen transformer type is modelled for 0.63 MVA, your sgen has p_mw=2. The transformer type is undersized.

Greetz Jan

Joboreh97, 2023-04-24:
@jwiemer112 

Thanks for your response. 

1) In pandapower design, does the transformer in an electrical network supply current to the generator in the same network by default? I want the generator to funtion as an independent power supply,thereby reducing the workload of the transformer. But I am not so sure how this should be implemented in a network. I would not want the transformer to supply current to the generator

2) Or is it the norm in electrical power networks, that the generator power must be lower than that of the transformer power?

Could you please explain to me,why it is important that the transformer power has to be higher than the generator power?

Thanks in advance

jwiemer112, 2023-04-24:
Hey, 

a generator creates feed in to a network. If the feed in isn't consumed locally by loads the power flow will go through lines and will be balanced by the external grid. If there is a transformer modelled in the grid and the power flow goes through it, it has to be capable to withstand those currents.

A transformer doesn't supply, it transforms voltages to a higher or lower voltage level. 

I do not fully understand what your goal is, feel free to post a minimal example of your grid and what your goal is.

Checkout our documentation: https://pandapower.readthedocs.io/en/latest/about.html

Greetz Jan

Joboreh97, 2023-04-25:
I have sent a private message to one of your social media network with a description of the task. I’m really looking forward to your reply.

Thanks

From: Jan Wiemer
Sent: Monday, 24 April 2023 12:38
To: e2nIEE/pandapower
Cc: Joboreh97; Mention
Subject: Re: [e2nIEE/pandapower] Unexpected plot result when modelling a chpas sgen to reduce the loading of transformer (Issue #1997)

Hey,
a generator creates feed in to a network. If the feed in isn't consumed locally by loads the power flow will go through lines and will be balanced by the external grid. If there is a transformer modelled in the grid and the power flow goes through it, it has to be capable to withstand those currents.
A transformer doesn't supply, it transforms voltages to a higher or lower voltage level.
I do not fully understand what your goal is, feel free to post a minimal example of your grid and what your goal is.
Checkout our documentation: https://pandapower.readthedocs.io/en/latest/about.html
Greetz Jan

jwiemer112, 2023-04-28:
We solved the problems in a private Meeting.

closed 2023-04-28 by jwiemer112

Joboreh97, 2023-04-28:
Thanks so much Jan. I really really appreciate your help and detailed explanation for every step