e2nIEE/pandapower #2003: IEEE 13 case power flow

rhythm232, 2023-04-29:
Hi
How to use pandapower to perform power flow calculation on IEEE 13 case?
# define unsymmetric line as standard type (component library)
line_601 = {"r_ohm_per_km":  0.4060, "x_ohm_per_km": 0.3895, "c_nf_per_km": 0.012029,  "r0_ohm_per_km":  0.4060,"x0_ohm_per_km": 1.2036,"c0_nf_per_km":  0.0054415,"max_i_ka": 0.5}
pp.create_std_type(My_13case,line_601, name="line_601",element = "line")

# create lines
line_r650_632 = pp.create_line(My_13case,from_bus=b650, to_bus=b632, length_km=0.6096, name="Line_r650_632",std_type="line_601",type="ol",index="r650632") 

# create trafos
t1 = pp.create_transformer(My_13case, hv_bus=b633, lv_bus=b634, std_type="unsymmetric_trafo", name="Trafo")
# create network feeder
r650 = pp.create_ext_grid(My_13case, bus=b650, vm_pu=1.05, p_mw=0.8, min_p_mw=0.05, max_p_mw=0.8, 
                          q_mw=0.5, min_q_mw=0, max_q_mw=0.5,name="Grid Connection", s_sc_max_mva=1000, rx_max=1, x0x_max=1, r0x0_max=1)

but it seems doesn t work

could someone help me?
Thank you

### Feature Checklist
* [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests
* [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [ ] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

**Describe the solution you'd like**
A clear and concise description of what you want to happen. You can also add some mock-up code for a desired behavior.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here. Please add related issues or "discussions" if there are any.

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `controller`, `OPF`...)

edxu96, 2023-04-30:
Hi, @rhythm232. As I mentioned in #836, the less-than-3-phase line hasn't been implemented in `pandapower`. Maybe one of the maintainers can confirm again.

You can use [OpenDSS](https://www.epri.com/pages/sa/opendss) for unbalanced ACPF with less-than-3-phase lines.

rhythm232, 2023-05-01:
Hi Edward Xu

Thank you for your reply.
I used Opendss for unbalanced ACPF. However it seems cant use for OPF.
Do you know any method could get the result of 3-phase OPF like IEEE 13?

rhythm232

edxu96, 2023-05-01:
You're welcome. You can see if https://github.com/lanl-ansi/PowerModelsDistribution.jl works for you

rbolgaryn, 2023-07-19:
it is correct, at the moment it is not supported in pandapower.

closed 2023-07-19 by rbolgaryn