e2nIEE/pandapower #2009: [bug] YOUR TITLE

viktor-ktorvi, 2023-05-06:
### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [x] Tried basic troubleshooting (if a bug/error)
    * [x] Restarted interpreter
    * [x] Checked the pythonpath
* [x] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description
When converting a net to a PPC and running OPF in pypower I get the following error(see traceback).

### What steps reproduce the problem?

Here's the simple code that generates the error:

```python
import pandapower.networks as pn

from pandapower.converter.pypower import to_ppc

from pypower.runpf import runpf
from pypower.runopf import runopf
from pypower.ppoption import ppoption

if __name__ == "__main__":
    net = pn.create_kerber_dorfnetz()
    ppc = to_ppc(net, init="flat", mode="opf")

    ppopt = ppoption(OUT_ALL=0, VERBOSE=0)
    ppc_pf, converged_pf = runpf(ppc, ppopt=ppopt)
    ppc_opf, converged_opf = runopf(ppc, ppopt=ppopt)
```

### What is the expected output? What do you see instead?

I expected there to be no issue. I suspect that something with the `to_ppc` isn't creating a correct PPC object.

### Paste Traceback/Error Below (if applicable)

```python-traceback

    ppc_opf, converged_opf = runopf(ppc, ppopt=ppopt)
  File "C:\Users\viktor\anaconda3\envs\eongnn\lib\site-packages\pypower\runopf.py", line 31, in runopf
    r = opf(casedata, ppopt)
  File "C:\Users\viktor\anaconda3\envs\eongnn\lib\site-packages\pypower\opf.py", line 172, in opf
    results, success, raw = opf_execute(om, ppopt)
  File "C:\Users\viktor\anaconda3\envs\eongnn\lib\site-packages\pypower\opf_execute.py", line 81, in opf_execute
    results, success, raw = pipsopf_solver(om, ppopt)
  File "C:\Users\viktor\anaconda3\envs\eongnn\lib\site-packages\pypower\pipsopf_solver.py", line 134, in pipsopf_solver
    solution = pips(f_fcn, x0, A, l, u, xmin, xmax, gh_fcn, hess_fcn, opt)
  File "C:\Users\viktor\anaconda3\envs\eongnn\lib\site-packages\pypower\pips.py", line 266, in pips
    h = hn if Ai is None else r_[hn, Ai * x - bi] # inequality constraints
  File "C:\Users\viktor\anaconda3\envs\eongnn\lib\site-packages\numpy\lib\index_tricks.py", line 412, in __getitem__
    res = self.concatenate(tuple(objs), axis=axis)
  File "<__array_function__ internals>", line 180, in concatenate
ValueError: all the input arrays must have same number of dimensions, but the array at index 0 has 2 dimension(s) and the array at index 1 has 1 dimension(s)

Process finished with exit code 1

```

## Versions

* pandas version: 1.5.3
* networkx version: 3.0
* scipy version: 1.6.1
* numpy version:  1.22.0
* packaging version:  23.0
* tqdm version:  4.65.0
* deepdiff version:  6.3.0
* Operating System name/version: Windows 10

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

vogt31337, 2024-11-21:
@viktor-ktorvi is this still happening?

viktor-ktorvi, 2024-11-22:
Best to check the code snippet that should recreate the error.

closed 2025-11-25 by vogt31337