e2nIEE/pandapower #2041: [bug] Incompatibility with numba 0.57.0 version information

meyer89, 2023-05-15:
### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [x] Tried basic troubleshooting (if a bug/error)
    * [x] Restarted interpreter
    * [x] Checked the pythonpath
* [x] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description

Pandapower is not compatible with numba 0.57.0. Installing the older numba version 0.56.4 solves the problem.

The interface changed in numba and the attribute `version_version` is no longer available.

### What steps reproduce the problem?

Use fresh virtual environment or docker image.

1. Execute `pip install numba==0.57.0`
2. Execute `pip install pandapower`
3. Execute `python -c "from pandapower.auxiliary import _check_if_numba_is_installed; print(_check_if_numba_is_installed(True))"` 

### What is the expected output? What do you see instead?

Expected output: 
> True

Actually observed output: 
> numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)

> False

### Paste Traceback/Error Below (if applicable)

```python-traceback
>>> from numba._version import version_version as numba_version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'version_version' from 'numba._version' (/usr/local/lib/python3.10/site-packages/numba/_version.py)
```

Old content of numba._versions.py

```python
version_version = '0.56.4'
version_full = '288a38bbd5a15418a211bf067878dfdf3c139509'
def get_versions(default={}, verbose=False):
    return {'version': version_version, 'full': version_full}
```

New content

```python
import json

version_json = '''
{
 "date": "2023-05-01T10:59:47-0500",
 "dirty": false,
 "error": null,
 "full-revisionid": "4fd4e39c672d119b54a2276d170f270764d2bce7",
 "version": "0.57.0"
}
'''  # END VERSION_JSON

def get_versions():
    return json.loads(version_json)
```

Importing the function `get_versions()` and evaluation the dictionary key `version` would work in both numba versions.

## Versions

* pandas version: 2.0.1
* networkx version: 3.1
* scipy version: 1.10.1
* numpy version: 1.24.3
* packaging version: 23.1
* tqdm version: 4.65.0
* deepdiff version: 6.3.0
* Operating System name/version: Docker-Image python:3.10 @ 2b8b079d7548

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

SteffenMeinecke, 2023-05-26:
With this set of packages I get the a NumbaDeprecationWarning:
```
C:\Users\smeinecke\Documents\GIT\pandapower\pandapower\pypower\makeLODF.py:25: NumbaDeprecationWarning: The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.
  def update_LODF_diag(LODF): # pragma: no cover
```
Maybe @rbolgaryn, you can have a look since it it about LODF?

deepdiff                      5.8.1
matplotlib                    3.7.1
networkx                      2.6.3
numba                         0.57.0
numpy                         1.23.5
packaging                     23.0
pandas                        2.0.1
scipy                         1.10.1

AnkurArohi, 2023-06-05:
I have encountered the same problem, there is a way out of it although.

The auxiliary script in pandapower is ending up in exception because the way the script is trying to extract the numba version, is not compatible with the numba _version script. Hence the output of the 

 from numba._version import version_version as numba_version

leads already to exception.

I will try to create a PR.

CarliJoy, 2023-06-27:
Is this bug fixed?

CarliJoy, 2023-06-27:
Btw. instead of using private imports of modules that are likely to change to determine the version of modules, I would recommend to use `importlib.metadata.version("numba")`.

This work from Python 3.8 onwards.

See https://stackoverflow.com/questions/20180543/how-do-i-check-the-versions-of-python-modules

closed 2023-07-12 by rbolgaryn

RedBatProject, 2023-08-27:
it still didnt work for me, i have python 3.10 and numba 0.57

SteffenMeinecke, 2023-09-11:
sorry to say @RedBatProject but reckon it is something with your python installation, since it is working with python 3.10 and numba 0.57 on all laptops I know.

brauliobarahona, 2023-11-03:
hallo everybody, I observed the same issue as @RedBatProject :

```bash
 python3.10 -c "from pandapower.auxiliary import _check_if_numba_is_installed; print(_check_if_numba_is_installed(True))"
numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)

False
```

I am running on macOS Sonoma, python 3.10, pandapower 2.13.1, numba 0.57

```
Package         Version
--------------- ------------
deepdiff        6.6.1
llvmlite        0.40.1
networkx        3.2.1
numba           0.57.0
numpy           1.24.4
ordered-set     4.1.0
packaging       23.2
pandapower      2.13.1
pandas          2.1.2
pip             23.3.1
python-dateutil 2.8.2
pytz            2023.3.post1
scipy           1.11.3
setuptools      65.5.0
six             1.16.0
tqdm            4.66.1
tzdata          2023.3
```