e2nIEE/pandapower #2060: Question on recycle for 3ph power flow

hakanozlemis, 2023-06-01:
Hello dear pandapower community,

I had a question to you regarding the time series simulation performance for 3ph power flow. In the meanwhile, I have faced a significant performance drop (10-times slower) after changing from single-phase to three-phase simulation which is not only due to problem complexitiy arising with three phases. I have realized then that the `recycle` feature is not available for 3ph-simulations (`runpp_3ph` and asymmetric elements such as `asymmetric_load`), see the following code lines:
- https://github.com/e2nIEE/pandapower/blob/cffe13487f0cd94823328f3a6a738a4e8b604289/pandapower/timeseries/run_time_series.py#L272
- https://github.com/e2nIEE/pandapower/blob/9a975c73494e0fe9d1c16439fef098c0be0d5942/pandapower/control/controller/const_control.py#L88

I got curious about if you have already a workaround solution to use recycling also for 3ph-simulations or are planning to add this feature as well. I would be glad to hear back from you.

You can find my both test cases below.

Best,
Hakan Özlemis

---

Single-phase timeseries simulation:
```
import simbench as sb
import pandapower as pp

# Simbench network
net_name = "1-LV-semiurb4--2-no_sw"
net = sb.get_simbench_net(net_name)

# Simbench profiles, !!! sets the controllers recycle=True already
sb.apply_const_controllers(net, sb.get_absolute_values(net, True))

# Use C++ based NR solver
pp.set_user_pf_options(net, lightsim2grid=True)

# Initialize result tables
pp.runpp(net)

# Run 3ph timeseries simulation for 1000 steps 
pp.timeseries.run_timeseries(net, time_steps=range(1000), verbose=True, run=pp.runpp)
```

Three-phase timeseries simulation:
```
import simbench as sb
import pandapower as pp

# Simbench network
net_name = "1-LV-semiurb4--2-no_sw"
net = sb.get_simbench_net(net_name)

# Simbench profiles, !!! sets the controllers recycle=True already but neglected due to runpp_3ph
sb.apply_const_controllers(net, sb.get_absolute_values(net, True))

# Prepare network for 3ph power flow
net.ext_grid['s_sc_max_mva'] = 10000
net.ext_grid['rx_max'] = 0.1
net.ext_grid['r0x0_max'] = 0.1
net.ext_grid['x0x_max'] = 1.0

net.line['c0_nf_per_km'] = 0
net.line['r0_ohm_per_km'] = 5 * net.line['r_ohm_per_km']
net.line['x0_ohm_per_km'] = 3 * net.line['x_ohm_per_km']

net.trafo['vector_group'] = 'Dyn'
net.trafo["vk0_percent"] = net.trafo["vk_percent"]
net.trafo["vkr0_percent"] = net.trafo["vkr_percent"]
net.trafo["mag0_percent"] = 1
net.trafo["mag0_rx"] = 0
net.trafo["si0_hv_partial"] = 0.9
net.trafo.pfe_kw = net.trafo.pfe_kw / 3
net.trafo.i0_percent = net.trafo.i0_percent / 3

# Use C++ based NR solver
pp.set_user_pf_options(net, lightsim2grid=True)

# Initialize result tables
pp.runpp(net)

# Run 3ph timeseries simulation for 1000 steps 
pp.timeseries.run_timeseries(net, time_steps=range(1000), verbose=True, run=pp.runpp_3ph)
```

hakanozlemis, 2023-06-23:
@SteffenMeinecke, @rbolgaryn, any suggestions for this topic?

rbolgaryn, 2023-06-27:
Dear @hakanozlemis ,

at the moment we do not have an immediate plan to invest effort in the improvement of the performance of 3ph power flow. 

If you are interested in implementing the recycle feature, we will review your pull request and give our comments, and answer specific questions.

Would you be interested in implementing the recycle feature for the 3ph power flow?

Best regards,
Roman

rbolgaryn, 2023-06-27:
There is an implementation of power grid model interface that is faster.

Nevertheless, it would be nice to have a recycle feature implemented in pandapower.

hakanozlemis, 2023-06-27:
Dear @rbolgaryn,

Thanks for your quick reply!

I am currently writing my master's thesis and I wanted accelerate the simulations for my thesis. However, I am racing quite against the clock, as you can understand.
I am unfortunately not expert in source code of Pandapower, thus, I cannot estimate how much work it would be. Could you please provide me an estimate of how much you think the implementation would take?

I had a follow-up question to your second message. You probably mean LightSim2Grid, right? I am already using it, but I would appreciate if you can let me know if there is an alternative. However, I think, the missing recycling is the limiting factor at this point.

Best,
Hakan Özlemis

rbolgaryn, 2023-06-29:
Dear @hakanozlemis ,

here is what I meant:

https://github.com/e2nIEE/pandapower/blob/4e04240697bea78b9a5e3b757d0a0a5233efe36e/pandapower/run.py#L253

As for the effort, I think between days and weeks.

Best regards,
Roman

hakanozlemis, 2023-07-03:
Dear @rbolgaryn ,

I did not know about the cool PGM extension. Thanks for letting me know about it!

Just to document it further regarding the solver: In my case, the limiting factor still stays the preparation (e.g., calulcation of Ybus) for numerical computation instead of the numerics or the asymmetric solver itself.

Thanks also for letting me know about your effort estimate. Just so you know, I do not think I cannot afford it in near future (6 months), but we can see if we manage something after that!

Best regards,
Hakan

vogt31337, 2024-11-21:
@hakanozlemis what is the progress on this topic? A PR would be appreciated.

hakanozlemis, 2024-12-20:
@vogt31337 I am not working with pandapower for some time and I cannot provide any support on this, sorry.