e2nIEE/pandapower #2062: [bug] with vector_group in create_transformer_from_parameters() latest 2.13.1

mul923, 2023-06-04:
### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [x] Tried basic troubleshooting (if a bug/error)
    * [x] Restarted interpreter
    * [x] Checked the pythonpath
* [x] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description
I have the following error setting vector_group in the create_transformer call
ValueError: could not convert string to float: 'Dyn'
It looks like it is related to recent mods to pandapower/create.py line 2841

### What steps reproduce the problem?

Simple net setup with just transformer will trigger

import pandapower as pp

net = pp.create_empty_network()
bus1 = pp.create_bus(net, vn_kv=20, name="Bus 1")
bus2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")

trafo = pp.create_transformer_from_parameters(net, hv_bus=bus1, lv_bus=bus2, sn_mva=30,
        vn_lv_kv=0.4, vn_hv_kv=20, vkr_percent=1.58, vk_percent=4,
        pfe_kw=0.8, i0_percent=0.32, vector_group='Dyn',
        vk0_percent=4, vkr0_percent=0.158, mag0_percent=50,
        mag0_rx=0.1, tap_phase_shifter=False,
        si0_hv_partial=0.9, in_service=True, shift_degree=30)

### What is the expected output? What do you see instead?
expect no errors. See ValueError: could not convert string to float: 'Dyn' related to vector_group

Adding back in old code: 
_create_column_and_set_value(net, index, vector_group, "vector_group", "trafo", dtyp=str, default_val=None)
and that function seems to fix

### Paste Traceback/Error Below (if applicable)

```python-traceback

Traceback (most recent call last):
  File "testpp.py", line 7, in <module>
    trafo = pp.create_transformer_from_parameters(net, hv_bus=bus1, lv_bus=bus2, sn_mva=30,
  File ".../.local/lib/python3.8/site-packages/pandapower/create.py", line 2841, in create_transformer_from_parameters
    _set_value_if_not_nan(net, index, vector_group, "vector_group", "trafo", dtype=str)
  File ".../.local/lib/python3.8/site-packages/pandapower/create.py", line 4815, in _set_value_if_not_nan
    net[element_type].at[index, column] = value
  File ".../.local/lib/python3.8/site-packages/pandas/core/indexing.py", line 2286, in __setitem__
    return super().__setitem__(key, value)
  File ".../.local/lib/python3.8/site-packages/pandas/core/indexing.py", line 2237, in __setitem__
    self.obj._set_value(*key, value=value, takeable=self._takeable)
  File ".../.local/lib/python3.8/site-packages/pandas/core/frame.py", line 3826, in _set_value
    series._values[loc] = value
ValueError: could not convert string to float: 'Dyn'

```

## Versions

* pandas version: 1.3.3
* networkx version: 2.7.1
* scipy version: 1.8.1
* numpy version: 1.20.3
* packaging version: 22.2.2
* tqdm version: 4.64.0
* deepdiff version: 6.2.3
* Operating System name/version: Ubuntu 18.04.6 LTS

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

jwiemer112, 2023-06-15:
Hey @SteffenMeinecke,

can you have a look at this?

SteffenMeinecke, 2023-06-21:
Sorry @mul923 I couldn't reproduce you error. Do you have the error after upgrading the python packages?

rbolgaryn, 2023-07-19:
Hi @mul923 is the issue still relevant? Have you tried the suggestion of @SteffenMeinecke to update all your Python packages?

mul923, 2023-07-19:
Hi @SteffenMeinecke and @rbolgaryn, Issue can be closed I think as updating to later libs seems to resolve.  Now using pandas 2.0.3 and all ok with latest pandapower.  One other issue with numba 0.57 but I don't think related to this.  Using numba 0.56.4 and all ok.

closed 2023-07-19 by SteffenMeinecke