e2nIEE/pandapower #2068: [feature] using parallel threading from PowerGridModel for contingency analysis

AnkurArohi, 2023-06-07:
### Feature Checklist
* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests
* [x ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

**Describe the solution you'd like**
As the run power flow is made faster considering the PowerGridModel module as the solver, similarly one could imagine using PowerGridModel solver for executing contingency analysis, the advantage being that the PowerGridModel has a feature of parallel thread based computation which in principle should also improve the performance.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here. Please add related issues or "discussions" if there are any.

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `controller`, `OPF`...)

rbolgaryn, 2023-07-03:
I have been working on this PR to implement contingency analysis with loghtsim2grid, but I haven't finished it yet. You can see if this is useful for you:

https://github.com/e2nIEE/pandapower/pull/1837

AnkurArohi, 2023-07-05:
@rbolgaryn Yes sure I will have a look

closed 2023-07-19 by rbolgaryn