e2nIEE/pandapower #2075: "va_degree" after load flow in low voltage distribution networks is giving very high negative values(in the range of -148degrees))

rahatN-tech, 2023-06-27:
what is the physical significance of getting such large angles after load flow in the networks  below 20kV. Can such values lead to failed state estimation.

rbolgaryn, 2023-07-03:
Hi @rahatN-tech 

did you calculate power flow with the parameter calculate_voltage_angles = True?

Are there any transformers with angle shift? If so, please refer to transformer documentation.

Best regards,
Roman

rbolgaryn, 2023-07-12:
Hi @rahatN-tech 

is the issue still relevant?

rbolgaryn, 2023-07-19:
Closing due to inactivity. Feel free to reopen if still relevant.

closed 2023-07-19 by rbolgaryn

EeToSe, 2024-08-03:
> Hi @rahatN-tech
> 
> did you calculate power flow with the parameter calculate_voltage_angles = True?
> 
> Are there any transformers with angle shift? If so, please refer to transformer documentation.
> 
> Best regards, Roman

Hi @rbolgaryn，
I am facing a similar situation when I run powerflow.ipynb in tutorials. I followed the exact same code. The results return by `net.res_bus` after running pp.runpp(net): 
[image]

If I run 'pp.runpp(net, calculate_voltage_angles = False)', the results return by `net.res_bus`:
[image]

Could you please explain why this may happen?