e2nIEE/pandapower #2083: Run PowerFlow with NumBa

maxiackner, 2023-07-03:
Hi pandapower team, 
Really enjoying using your package, it's really great. I have a question concerning NumBa. I create a LV network and wanted to run a power flow calculation with runpp. However, when i do that, if works, but i get the warning: 

"numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)"

I am using python 3.8.16, pandapower version 2.13.1, and numba version 0.57.1. 
Do you know what the problem could be here? Looking forward for your response. 
[image]

rbolgaryn, 2023-07-03:
Hi @maxiackner ,

please check the list of installed packages. Make sure you are using a correct Python environment.

Best regards,
Roman

maxiackner, 2023-07-03:
I am using a Python environment with the following packages: 
[Packages.txt](https://github.com/e2nIEE/pandapower/files/11940217/Packages.txt)

rbolgaryn, 2023-07-03:
you can find where in the code numba is imported and where the ImportError is handled, then debug there and check which error actually occurs - and post the error message here

rbolgaryn, 2023-07-03:
> I am using a Python environment with the following packages: [Packages.txt](https://github.com/e2nIEE/pandapower/files/11940217/Packages.txt)

please paste the contents in the message here

maxiackner, 2023-07-03:
(master_thesis) C:\Users\Maximilian Lackner\OneDrive - ETH Zürich\ETH\Master\Master Thesis\Code> cmd /C ""C:\Users\Maximilian Lackner\.conda\envs\master_thesis\python.exe" "c:\Users\Maximilian Lackner\.vscode\extensions\ms-python.python-2023.10.1\pythonFiles\lib\python\debugpy\adapter/../..\debugpy\launcher" 60710 -- "C:\Users\Maximilian Lackner\OneDrive - ETH Zürich\ETH\Master\Master Thesis\Code\NG_network_generation.ipynb" "
Traceback (most recent call last):
  File "C:\Users\Maximilian Lackner\.conda\envs\master_thesis\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Maximilian Lackner\.conda\envs\master_thesis\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\Maximilian Lackner\.vscode\extensions\ms-python.python-2023.10.1\pythonFiles\lib\python\debugpy\adapter/../..\debugpy\launcher/../..\debugpy\__main__.py", line 39, in <module>
npy.py", line 135, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "c:\Users\Maximilian Lackner\.vscode\extensions\ms-python.python-2023.10.1\pythonFiles\lib\python\debugpy\_vendored\pydevd\_pydevd_bundle\pydevd_runpy.py", line 124, in _run_code
    exec(code, run_globals)
  File "C:\Users\Maximilian Lackner\OneDrive - ETH Zürich\ETH\Master\Master Thesis\Code\NG_network_generation.ipynb", line 135, in <module>
    "execution_count": null,
NameError: name 'null' is not defined

rbolgaryn, 2023-07-03:
You can check if this pull request is fixing your issue:

https://github.com/e2nIEE/pandapower/pull/2077

You can check out the branch from the pull request and see if it fixes the issue.

maxiackner, 2023-07-03:
> You can check if this pull request is fixing your issue:
> 
> #2077
> 
> You can check out the branch from the pull request and see if it fixes the issue.

How exactly can I let my code run with this branch from the pull request?

rbolgaryn, 2023-07-03:
You can follow this documentation (section development version): 

http://www.pandapower.org/start/#install

Except instead of the branch e2nIEE:develop you need to use CarliJoy:fix_numba_version_import

maxiackner, 2023-07-04:
> You can follow this documentation (section development version):
> 
> http://www.pandapower.org/start/#install
> 
> Except instead of the branch e2nIEE:develop you need to use CarliJoy:fix_numba_version_import

This worked. I don't get the warning anymore. However, I don't really see any performance improvements.

maxiackner, 2023-07-04:
For a annual powerflow calculation with 8760 timesteps, it takes about 2 minutes to calculate. Is this the typical duration with numba or should it be quicker?

rbolgaryn, 2023-07-04:
Do you still get the warning?

rbolgaryn, 2023-07-04:
You can speed up further by installing lightsim2grid:

pip install lightsim2grid

maxiackner, 2023-07-04:
> Do you still get the warning?

No, the warning has gone!

maxiackner, 2023-07-04:
> You can speed up further by installing lightsim2grid:
> 
> pip install lightsim2grid

I have installed it already. Do i have to activate it somehow? 

pp.runpp(lv_network, numba=True, lightsim2grid=True) Is this correct?

jiangce, 2023-07-06:
Hello everyone, I also noticed this issue, and I have checked the source code. It seems that the error is caused by importing numba_version after updating numba. I made the following modifications in auxiliary.py and it fixed the problem. The changes were made around line 44:

try:
    from numba import jit
    from numba import version_info
    numba_version = version_info.string
except ImportError:
    from .pf.no_numba import jit

rbolgaryn, 2023-07-06:
We already have a pending pull request that fixes the issue.

For lightsim2grid, if you installed it, it will be used by default.