e2nIEE/pandapower #2089: Result output for three phase powerflow

emily-ninestein, 2023-07-11:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hi there! I'm trying to run a three-phase powerflow (3ph_runpp) on the IEEE European LV network, but I am getting empty dataframes for the asymmetric load data. Is there something I'm missing? If I run a standard powerflow (runpp), I do get asymmetric load results, but they're single phase. I thought the issue might be because the European LV network only has one loaded phase per three-phase load (two of the phases are always 0), so I tried adding a load to each 0 phase, but I am still getting single-phase results. Here is the code I'm running:

`import pandapower as pp
import pandapower.networks as pn
from pandapower.plotting.plotly import pf_res_plotly 

def load_all_phases(net):
    '''adds a load to every phase that is currently at 0'''
    load_data_frame = net.asymmetric_load
    new_value = 0.000114 
    load_data_frame.replace(to_replace=0, value=new_value, inplace=True) # replace all 0s with new value
    net.asymmetric_load = load_data_frame
    return net

def main():
    net = pn.ieee_european_lv_asymmetric('on_peak_566') # create network
    net = load_all_phases(net) # add load wherever the load is 0

    print("Asymmetric load table:") # print the load data before the powerflow
    print(net.asymmetric_load) 

    pp.runpp_3ph(net) # run the three-phase powerflow
    # pp.runpp(net) # run standard powerflow

    print("Result asymmetric load table:") # print the result loads of the powerflow
    print(net.res_asymmetric_load)

main()`

Do you know why I am getting these results? Thanks!

### Label

- [ ] Relevant labels are selected

rbolgaryn, 2023-07-12:
Please use a different issue template where you can input code snippets for the minimal example

emily-ninestein, 2023-07-12:
@rbolgaryn Sure! Do you want me to close this issue and create a new one as a bug report? Or would a permalink to the code be better?

rbolgaryn, 2023-07-19:
Hi @emily-ninestein ,

no it is not necessary. I tried running your code and it seems like the power flow calculation does not converge. Is that the problem?

emily-ninestein, 2023-07-21:
@rbolgaryn I do not get an error message saying that it did not converge, but maybe that could be the underlying issue? The issue I am having is that the result dataframe for the asymmetric load comes up empty when the runpp_3ph function is run. However, when the standard runpp function is run, the asymmetric load result dataframe is filled, but it only returns two parameters (p_mw and q_mvar) instead of the 6 that are expected (p_a_mw, p_b_mw, p_c_mw, q_a_mvar, q_b_mvar, q_c_mvar). How can I get these 6 parameters the dataframe is supposed to have according to the documentation? 
[image]

rbolgaryn, 2023-07-25:
The 6 parameters are only returned when the calculation is unbalanced power flow. In my case, it did not converge - this could be the reason why the dataframe is empty.

rbolgaryn, 2023-07-25:
you need to use the function runpp_3ph

vogt31337, 2024-11-21:
@emily-ninestein is this still relevant?