e2nIEE/pandapower #2102: [bug] Numba cannot be imported in virtual environment in WSL2

shanks847, 2023-08-01:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
import numba #just to see if this would help
net = pp.create_empty_network()
b1 = pp.create_bus(net, vn_kv=20.)
b2 = pp.create_bus(net, vn_kv=20.)
pp.create_line(net, from_bus=b1, to_bus=b2, length_km=2.5, std_type="NAYY 4x50 SE")
pp.create_ext_grid(net, bus=b1)
pp.create_load(net, bus=b2, p_mw=1.)

pp.runpp(net)

print(net.res_bus.vm_pu)
print(net.res_line.loading_percent)
```

### Issue Description and Traceback

When running, it states, numba cannot be imported and numba functions are disabled. I tried using a different virtual environment with conda and also using pyhthon -m pip install [pkg name] to install pandapower and numba. I'm still getting the error. I saw there has been previous issues in 2017 with this, and I would like to know if the issue has been resolved or if there is a possible fix.

### Expected Behavior

I would like for pandapower to access my installed version of Numba and use it for computation.

### Installed Versions

commit           : 0f437949513225922d851e9581723d82120684a6
python           : 3.11.4.final.0
python-bits      : 64
OS               : Linux
OS-release       : 5.15.90.1-microsoft-standard-WSL2
Version          : #1 SMP Fri Jan 27 02:56:13 UTC 2023
machine          : x86_64
processor        : x86_64
byteorder        : little
LC_ALL           : None
LANG             : C.UTF-8
LOCALE           : en_US.UTF-8

pandas           : 2.0.3
numpy            : 1.24.3
pytz             : 2023.3
dateutil         : 2.8.2
setuptools       : 68.0.0
pip              : 23.2.1
Cython           : None
pytest           : None
hypothesis       : None
sphinx           : None
blosc            : None
feather          : None
xlsxwriter       : None
lxml.etree       : None
html5lib         : None
pymysql          : None
psycopg2         : None
jinja2           : None
IPython          : None
pandas_datareader: None
bs4              : None
bottleneck       : None
brotli           : None
fastparquet      : None
fsspec           : None
gcsfs            : None
matplotlib       : None
numba            : 0.57.0
numexpr          : None
odfpy            : None
openpyxl         : None
pandas_gbq       : None
pyarrow          : None
pyreadstat       : None
pyxlsb           : None
s3fs             : None
scipy            : 1.11.1
snappy           : None
sqlalchemy       : None
tables           : None
tabulate         : None
xarray           : None
xlrd             : None
zstandard        : None
tzdata           : 2023.3
qtpy             : None
pyqt5            : None

### Label

- [ ] Relevant labels are selected

rbolgaryn, 2023-08-01:
Hi @shanks847 ,

what if you install pandapower from the develop branch?

```
pip install git+https://github.com/e2nIEE/pandapower.git@develop
```

shanks847, 2023-08-01:
That works, thank you. Would this issue be fixed in the next stable release of pandapower?

closed 2023-08-01 by shanks847

stibu81, 2023-08-18:
I have installed the develop branch as indicated by @rbolgaryn today and still get this warning. Am I doing something wrong or has this issue returned?