e2nIEE/pandapower #2105: [bug] Pandapower interferes with matplotlib savefig

chrisyeh96, 2023-08-08:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import matplotlib.pyplot as plt
import pandapower

fig, ax = plt.subplots()
ax.scatter(range(5), range(5))
fig.savefig('test.svg')
```

### Issue Description and Traceback

When pandapower is imported, matplotlib `savefig()` may run into a bug where the `GraphicsContextBase._capstyle` is set to a `str` instead of a `CapStyle` instance. Calling the proper `set_capstyle()` method solves this issue. Also, somehow, this issue does not arise when calling `fig.savefig('test.png')`. It only arises when the figure save type is SVG.

The following code works fine. Notice that I have commented out `import pandapower`:
```python
import matplotlib.pyplot as plt
# import pandapower

fig, ax = plt.subplots()
ax.scatter(range(5), range(5))
fig.savefig('test.svg')
```

However, if I uncomment the `import pandapower` line, then I will get an error:
```python
import matplotlib.pyplot as plt
import pandapower

fig, ax = plt.subplots()
ax.scatter(range(5), range(5))
fig.savefig('test.svg')
```

Error:

```
Traceback (most recent call last):
  File "/home/user/testenv/test.py", line 6, in <module>
    fig.savefig('test.svg')
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/figure.py", line 3378, in savefig
    self.canvas.print_figure(fname, **kwargs)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/backend_bases.py", line 2366, in print_figure
    result = print_method(
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/backend_bases.py", line 2232, in <lambda>
    print_method = functools.wraps(meth)(lambda *args, **kwargs: meth(
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/backends/backend_svg.py", line 1369, in print_svg
    self.figure.draw(renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/artist.py", line 95, in draw_wrapper
    result = draw(artist, renderer, *args, **kwargs)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
    return draw(artist, renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/figure.py", line 3175, in draw
    mimage._draw_list_compositing_images(
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
    return draw(artist, renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/axes/_base.py", line 3064, in draw
    mimage._draw_list_compositing_images(
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
    return draw(artist, renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/collections.py", line 972, in draw
    super().draw(renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
    return draw(artist, renderer)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/collections.py", line 405, in draw
    renderer.draw_markers(
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/backends/backend_svg.py", line 717, in draw_markers
    style = self._get_style_dict(gc, rgbFace)
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/backends/backend_svg.py", line 609, in _get_style_dict
    if gc.get_capstyle() != 'butt':
  File "/home/user/miniconda3/envs/testenv/lib/python3.10/site-packages/matplotlib/backend_bases.py", line 820, in get_capstyle
    return self._capstyle.name
AttributeError: 'str' object has no attribute 'name'
```

### Expected Behavior

I would expect the following 2 code blocks to produce identical (or at least similar) results:

```python
import matplotlib.pyplot as plt
# import pandapower

fig, ax = plt.subplots()
ax.scatter(range(5), range(5))
fig.savefig('test.svg')
```

and 

```python
import matplotlib.pyplot as plt
import pandapower

fig, ax = plt.subplots()
ax.scatter(range(5), range(5))
fig.savefig('test.svg')
```

The 1st code block works fine, whereas the 2nd code block throws an `AttributeError`.

### Installed Versions

OS: Ubuntu 22.04 LTS
Python 3.10
Matplotlib 3.7.2
Pandapower 2.13.1

### Label

- [X] Relevant labels are selected

rbolgaryn, 2023-08-09:
Dear @chrisyeh96 ,

I merged your pull request. Is the issue fixed?

closed 2023-08-09 by rbolgaryn

chrisyeh96, 2023-08-09:
Yes, the issue is fixed.

PR #2106 (2023-08-09): Fix plotting bug
Fixes #2105