e2nIEE/pandapower #2128: [feature] consider ext_grid dispatch power in distributed slack calculation

SteffenMeinecke, 2023-09-19:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [X] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

In a multi-generator grid, one of these should be modeled as slack generator or as ext_grid. If the slack should be distributed, ext_grids cannot participate anymore if all generating units should have a dispatch value to be considered. If, therefore, all slack participating units are modeled as generator (and one of theses with the flag `slack=True`), the option of setting `va_degree`, as possible with ext_grids, is gone.
This might be manageable, but it would be more convenient to not first need to replace the ext_grid to a slack generator.

### Feature Description

If the column `p_disp_mw` is given in `net.ext_grid` it is considered in distributed slack calculation as initial active power vector for ext_grids.
(as considered in the existing distributed slack controller)

### Label

- [X] Relevant labels are selected

rbolgaryn, 2023-09-25:
Why not just add a load or sgen at the ext grid bus?

vogt31337, 2024-11-21:
@SteffenMeinecke still relevant? Could you close the issue otherwise?

SteffenMeinecke, 2024-11-22:
I am still the opinion that this would be a nice feature and adding a load with negative value can just be a workaround.

vogt31337, 2026-11-11:
Solution add a load or sgen with the initial value. Closed due to irrelevance.

closed 2026-11-11 by vogt31337