e2nIEE/pandapower #2131: [bug] Unexplained difference in "branch" data for ieee118

BDonnot, 2023-09-21:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
import pandapower.networks as pn
case = pn.case118()

# run a powerflow to force the computation of the "branch" data
pp.runpp(case)

# retrieve the "branch" data
case["_ppc"]["internal"]["branch"][180, 2:5] # to retrieve the r, x and h value
# > array([0.00137296+0.j        , 0.01595947+0.j        , 0.63963231-0.00014115j])

# should be as far I understand:
# 0.00138, 0.016, 0.638 in pypower https://github.com/rwl/PYPOWER/blob/977e1a237c5dc4a6c2e8c5ad2038b0c4d5cd64e0/pypower/case118.py#L323C18-L323C39
# 0.00138    0.016    0.638 if you prefer https://matpower.org/docs/ref/matpower5.0/case118.html (line 0313)
```

### Issue Description and Traceback

As far as I understand (from the doc here https://pandapower.readthedocs.io/en/latest/networks/power_system_test_cases.html#case-118), the 118 grid "case118" should match the data coming from the ieee 118 dataset.

This dataset is given in "per unit" convention. Pandapower uses "engineering" units (ohm, siemens, etc.) so I expect  values in the `trafo` and `line` table of pandapower to be different to the value given in the litterature for the 118 (that are given in per unit)

But what I would have expected is that once converted back (internally)  to per unit values, this would give the parameters in the litterature (I checked and pypower has the same one as matpower which has the same one as the original cdf one).

Unfortunately when i looked at the "branch" data of the internal pandapower `case["_ppc"]["internal"]["branch"]` it does not match the original data (slight difference for some rows, I put one here but there 1 or 2 others)

### Expected Behavior

The "per united" values should match the one in the original file (as far as I understand)

### Installed Versions

* python version: Python 3.8.10 (default, Sep 28 2021, 16:10:42)
* pandas version:  2.0.3
* networkx version:   3.1
* scipy version:  1.10.1
* numpy version:  1.24.4
* packaging version: not usefull
* tqdm version:  not usefull
* deepdiff version:  not usefull
* Operating System name/version: ubuntu 20.04

### Label

- [ ] Relevant labels are selected

BDonnot, 2024-03-26:
Hello,

Does someone has any news about this issue ?

rbolgaryn, 2024-07-15:
Hi @BDonnot ,

Thank you for pointing this out!

We are looking closely at the G and B in pandapower calculations at the moment, we will take a look at this issue as well. I think we will be done with the G and B issue in a couple of weeks.

Best regards,
Roman

pawellytaev, 2024-08-01:
Hi @BDonnot thanks for bringing that up!
The difference is due to pandapower's assumption of a T-equivalent circuit for trafos, matpower/mpc formats assume Pi-equivalent circuit. So in the conversion pd2ppc pandapower does a Wye-Delta conversion to bring it into the Pi-equivalent circuit:
https://github.com/e2nIEE/pandapower/blob/ed0e8e534193927b6bfe0d5a31fd8014e7dc0cca/pandapower/build_branch.py#L325

selecting trafo_model='pi' in runpp gives the same results
`
import pandapower as pp
import pandapower.networks as pn
case = pn.case118()

# run a powerflow to force the computation of the "branch" data
pp.runpp(case, trafo_model='pi')

# retrieve the "branch" data
case["_ppc"]["internal"]["branch"][180, 2:5] # to retrieve the r, x and h value
# > array([0.00138+0.j, 0.016  +0.j, 0.638  +0.j])

# should be as far I understand:
# 0.00138, 0.016, 0.638 in pypower https://github.com/rwl/PYPOWER/blob/977e1a237c5dc4a6c2e8c5ad2038b0c4d5cd64e0/pypower/case118.py#L323C18-L323C39
# 0.00138    0.016    0.638 if you prefer https://matpower.org/docs/ref/matpower5.0/case118.html (line 0313)
`

rbolgaryn, 2024-08-01:
Thank you @pawellytaev !

closed 2024-08-01 by pawellytaev