e2nIEE/pandapower #2133: Convergence Issues

PB-vines, 2023-09-22:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hello, 

I have a network that runs a PF and OPF when imported and all line lengths are 1. When I update the line lengths to what they are supposed to be, the network does not converge. Could someone give advice regarding fixing this issue? I have no impedance values close to zero, as checked with the diagnostic tool. 

I do get the warning: 

"ComplexWarning: Casting complex values to real discards the imaginary part
  Sf = V[ branch[:, F_BUS].astype(int) ] * conj(Yf * V)  ## cplx pwr at "from" bus, p["u"]."

and 

"Casting complex values to real discards the imaginary part
  muSt[il] = \"

However, I have verified that all inputs required by PP are correct. I am running a time series simulation, and found another post where someone had a ComplexWarning, but no responses were helpful. I am not running a 3 phase power flow.

Thanks.

### Label

- [X] Relevant labels are selected

pawellytaev, 2024-09-04:
Hi, it's hard to assist in this case without having an example net with values. Can you provide the code that creates the net?

Is the warning you're describing thrown when running runpp?

vogt31337, 2024-11-21:
@PB-vines If I remember correctly the values in pandapower lines are "per km" so if you update the lengths pandapower will internally multiply everything. Therefore you MUST update the corresponding values, otherwise they will be wrong, explaining your results...

teddy607, 2025-02-05:
I have a similar question. Initially, I had created my own network with custom line lengths and parameters, but when I adjusted my line length to anything greater than 2km it failed to converge and the diagnostic function was not indicating why. 

I then backtracked to to using an "out of the box" network below, where I adjusted the line lengths. Again anything over ~2km fails to converge. When I looked at the code that builds the branches, it takes length_km and multiplies the x_ohm_per_km and r_ohm_per_km to formulate the branch impedance. So scaling the length of the line automatically scales the impedance parameters of the line. So why does the network fail to converge? 

`import pandapower as pp

# Create the network
net = pp.networks.case24_ieee_rts()

# Set line lengths to 3 km
net.line["length_km"] = 3

# Run the power flow
pp.runpp(net)`

Could someone please assist me with understanding why the above code fails and what the potential fix should be to have this network work with longer line lengths? I can then apply this knowledge to other custom networks. Is there not enough load generated for the increased impedance? 

Thank you!