e2nIEE/pandapower #2134: Each timeseries cost

Aliadavid, 2023-09-25:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Is it possible to store the Dispatch Cost for each time step using the output_writer?

### Label

- [ ] Relevant labels are selected

pawellytaev, 2024-09-04:
Hi @Aliadavid, When using OPF in timeseries, you can use the output writer during run_timeseries to track the res_cost. Here are relevant tutorials that might help:
https://github.com/e2nIEE/pandapower/blob/develop/tutorials/opf_basic.ipynb
https://github.com/e2nIEE/pandapower/blob/develop/tutorials/timeseries_opf.ipynb

PB-vines, 2024-11-21:
@Aliadavid I have the same question. @pawellytaev those tutorials do not show logging the cost at each indepent timestep. If I have 24 time steps, I want 24 cost data points.

vogt31337, 2024-11-21:
@PB-vines I am not sure, but this should be in the internal data structure of the output writer which you can also access afterwards. Quick hack would be to just run a for-loop.

vogt31337, 2025-03-10:
@PB-vines and @Aliadavid like @pawellytaev mentioned, just add res_cost to the output_writer.
For example ow.log_variable("df.res_cost") <- I do not know in which dataframe these costs are written, you have to look that up yourself.

closed 2025-03-10 by vogt31337