e2nIEE/pandapower #2143: Understanding the bus indices in the json format

SanPen, 2023-10-06:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hi folks, 

I'm writing a [GridCal backend for Grid2Op](https://github.com/SanPen/Grid2OpBackends) and I need to read your json format ([this file](https://github.com/e2nIEE/pandapower/files/12829364/test_case14.json.txt) particularly)

In the file, the line and transformer indices don't seem to match the bus indices or the bus names properly.

The [function](https://github.com/SanPen/Grid2OpBackends/blob/main/gridcalBackend.py) I'm using is this:

```python

def read_pandapower_file(filename: str) -> MultiCircuit:
    """
    Parse pandapower json file into a GridCal MultiCircuit object
    :param filename: file path
    :return: MultiCircuit object
    """
    grid = MultiCircuit()

    with open(filename) as f:
        data = json.load(f)
        data2 = data['_object']

        if type(data2) == str:  # because madness
            data2 = json.loads(data2)

    # buses
    bus_dict_by_name = dict()
    bus_dict_by_index = dict()
    for i, row in decode_panda_structre(obj=data2['bus']).iterrows():

        name = 'sub_{}'.format(i)
        vmin = row['min_vm_pu'] if 'min_vm_pu' in row else 0.8 * row['vn_kv']
        vmax = row['max_vm_pu'] if 'max_vm_pu' in row else 1.2 * row['vn_kv']
        bus = dev.Bus(idtag='',
                      code='',
                      name=name,
                      active=bool(row['in_service']),
                      vnom=row['vn_kv'],
                      vmin=vmin,
                      vmax=vmax
                      )
        bus_dict_by_name[name] = bus
        bus_dict_by_index[i] = bus
        grid.add_bus(bus)

    # loads
    for i, row in decode_panda_structre(obj=data2['load']).iterrows():
        bus = bus_dict_by_index[row['bus']]

        name = "load_{0}_{1}".format(row['bus'], i)

        grid.add_load(bus, dev.Load(idtag='',
                                    code='',
                                    name=name,
                                    active=bool(row['in_service']),
                                    P=row['p_mw'] * row['scaling'],
                                    Q=row['q_mvar'] * row['scaling']))

    # shunt
    for i, row in decode_panda_structre(obj=data2['shunt']).iterrows():
[... 114 lines cut ...]
|--------|-------------------|-------------------|-------------------|-----------------------|-----------------------|-----------------------|-----------------------|-----------------------|-----------------------|-------------------|-------------------|-------------------|-------------------|-------------------|
| BUS 1  | 6.02502905576822  | -4.99913160079803 | 0                 | 0                     | -1.02589745497019     | 0                     | 0                     | 0                     | 0                     | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 2  | -4.99913160079803 | 9.52132361081478  | -1.1350191923074  | -1.68603315061494     | -1.7011396670944      | 0                     | 0                     | 0                     | 0                     | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 3  | 0                 | -1.1350191923074  | 3.12099490223296  | -1.98597570992556     | 0                     | 0                     | 0                     | 0                     | 0                     | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 4  | 0                 | -1.68603315061494 | -1.98597570992556 | 10.5129895220362      | -6.84098066149567     | 0                     | -2.33813727061847E-19 | 0                     | -3.3361493721743E-20  | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 5  | -1.02589745497019 | -1.7011396670944  | 0                 | -6.84098066149567     | 9.56801778356027      | -1.68932836094492E-19 | 0                     | 0                     | 0                     | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 6  | 0                 | 0                 | 0                 | 0                     | -1.68932836094492E-19 | 6.57992340746622      | 0                     | 0                     | 0                     | 0                 | -1.95502856317726 | -1.52596744045097 | -3.09892740383799 | 0                 |
| BUS 7  | 0                 | 0                 | 0                 | -2.33813727061847E-19 | 0                     | 0                     | 1.37724686538679E-18  | -3.22281001800826E-19 | -8.26296038519476E-19 | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 8  | 0                 | 0                 | 0                 | 0                     | 0                     | 0                     | -3.22281001800826E-19 | 3.22281001800826E-19  | 0                     | 0                 | 0                 | 0                 | 0                 | 0                 |
| BUS 9  | 0                 | 0                 | 0                 | -3.3361493721743E-20  | 0                     | 0                     | -8.26296038519476E-19 | 0                     | 5.32605503946736      | -3.90204955244743 | 0                 | 0                 | 0                 | -1.42400548701993 |
| BUS 10 | 0                 | 0                 | 0                 | 0                     | 0                     | 0                     | 0                     | 0                     | -3.90204955244743     | 5.78293430614783  | -1.8808847537004  | 0                 | 0                 | 0                 |
| BUS 11 | 0                 | 0                 | 0                 | 0                     | 0                     | -1.95502856317726     | 0                     | 0                     | 0                     | -1.8808847537004  | 3.83591331687766  | 0                 | 0                 | 0                 |
| BUS 12 | 0                 | 0                 | 0                 | 0                     | 0                     | -1.52596744045097     | 0                     | 0                     | 0                     | 0                 | 0                 | 4.01499202727289  | -2.48902458682192 | 0                 |
| BUS 13 | 0                 | 0                 | 0                 | 0                     | 0                     | -3.09892740383799     | 0                     | 0                     | 0                     | 0                 | 0                 | -2.48902458682192 | 6.72494614846623  | -1.13699415780633 |
| BUS 14 | 0                 | 0                 | 0                 | 0                     | 0                     | 0                     | 0                     | 0                     | -1.42400548701993     | 0                 | 0                 | 0                 | -1.13699415780633 | 2.56099964482626  |

Observe the index shuffling.

So my question is, how do you assign the bus indices and what is the relation to those in the loads, generators, etc..

Thanks!

### Label

- [ ] Relevant labels are selected

KS-HTK, 2023-10-09:
I assume you are attempting to avoid having a dependency on pandapower?
If that is not the case it would be far simpler to work with data from the pandas dataframes when they are loaded in pandapower.

The structure in the json matches the structure of the corresponding element in the pandapower documentation. From that you can see how the indeces are used.

From the Documentation of [net.line](https://pandapower.readthedocs.io/en/v2.13.1/elements/line.html#input-parameters):
Parameter|Datatype|Value Type|Explanation
-- | -- | -- | --
from_bus* | integer |   | Index of bus where the line starts
to_bus* | integer |   | Index of bus where the line ends

the integers stored there are an index of the `net.bus` dataframe.

For the [net.trafo](https://pandapower.readthedocs.io/en/v2.13.1/elements/trafo.html#input-parameters) table accordingly there exists:
Parameter | Datatype | Value Range | Explanation
-- | -- | -- | --
hv_bus* | integer |   | high voltage bus index of the transformer
lv_bus* | integer |   | low voltage bus index of the transformer

**There is no direct correlation of the index of a line or trafo to a index of a bus.**

Indices have nothing to do with the name of a bus. If the bus is named "Bus 0" that does not mean it has index 0. Sorting by name vs sorting by index thus may result in different sequences. Names are for better Identification by the user. They should not be used to match up the data from the different dataframes.

vogt31337, 2024-11-21:
@SanPen still working on this? Have you tried the suggestion of @KS-HTK ? If not could you close the issue?

SanPen, 2024-11-21:
I don't remember how this ended up, so I'm closing!

closed 2024-11-21 by SanPen