e2nIEE/pandapower #2146: [feature] CIM CGMES 3.0 to pandapower

MarcoPignati, 2023-10-09:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [X] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

CMGES 3.0 (based on CIM 17) is now part of the IEC 61970 standard. Utilities are suggested to upgrade from 2.4.15 to 3.0 (e..g, for better support for unbalanced network and other issues solved). I would like to know if pandapower is plannig to expand the cim2pp to support also the CGMES3.0 profile on top of the existing CGMES 2.4.15.

### Feature Description

I would like pandapower to be able to ingest CIM CGMES 3.0 files.  Check [interesting videos here](https://youtube.com/playlist?list=PLP7fkvyH8qQdxRE9fDDzlrsBo6l9xVWwU&si=p7GE9koLKRafLfXV), for changes between 2.4.15 and 3.0.0 

### Label

- [X] Relevant labels are selected

rbolgaryn, 2023-10-18:
Hi @MarcoPignati ,

we have tentative plans to implement CGMES 3.0 conversion to pandapower sometime next year.
We cannot confirm this with certainty at this moment, though.

Best regards,
Roman

MarcoPignati, 2023-10-19:
Hi, thanks for the answer. I'll stay tuned for updates. If we have resources, we'll see if we can contribute.

vogt31337, 2024-11-21:
afaik this is planned for one of the upcoming releases.

closed 2024-11-21 by vogt31337