e2nIEE/pandapower #2157: [bug] network is built and slack boolean is True in one generator but still warning about no reference

AnkurArohi, 2023-10-25:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
use CGMES importer 
net = from_cim.from_cim(file_list=file_list)
check if the slack boolean is set True
it is set True for one generator

While debugging i saw that the ppc for gen is an empty array , this leads to the issue 

Here is the exception 
    ref, _, _ = bustypes(ppc["bus"], ppc["gen"])
https://github.com/e2nIEE/pandapower/blob/35441514656b4124b2964a8111dbc575fdcf494d/pandapower/build_gen.py#L319C8-L319C8

The problem is that ppc["gen"] is empty
```

### Issue Description and Traceback

net.gen.loc[[30]].slack
Out[4]: 
30    True
Name: slack, dtype: bool

But there is an exception raised that reference is not set, this is because ppc array for gen is empty although there are gens in net

### Expected Behavior

power flow run

### Installed Versions

* python version: * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:
python           :   3.11.3.final.0
pandas           :   1.4.3
numpy            :   1.23.5
pandapower         2.13.1
scipy                     1.11.1
tqdm                      4.65.0
packaging                 23.1
deepdiff                  6.3.0 

os fedora

### Label

- [ ] Relevant labels are selected

mrifraunhofer, 2023-10-26:
Hallo Ankur,
vielen Dank für dein Feedback.
Du schreibst:
"net.gen.loc[[30]].slack
Out[4]:
30 True
Name: slack, dtype: bool

But there is an exception raised that reference is not set, this is because ppc array for gen is empty although there are gens in net"
Kannst du mal bitte prüfen, ob der "gen" ggf. nicht in_service ist? In unserem Test funktioniert der pandapower Lastfluss wenn es kein "ext_grid" aber einen "gen" mit "slack=True" und "in_service=True" gibt.
Gib uns bitte den Output von "net.gen.loc[[30]].in_service".

AnkurArohi, 2023-10-26:
gen is in service
I checked that already

AnkurArohi, 2023-10-26:
> Hallo Ankur, vielen Dank für dein Feedback. Du schreibst: "net.gen.loc[[30]].slack Out[4]: 30 True Name: slack, dtype: bool
>
> But there is an exception raised that reference is not set, this is [...]

Please write in english

mrifraunhofer, 2023-10-27:
Hey Ankur,

can you provide us the network or any further information to reproduce the error? Currently we can't reproduce the exception...

AnkurArohi, 2023-10-27:
Thanks for you help, yes I can provide you with further details in a teams call but not the network.

AnkurArohi, 2023-11-06:
Hello, we were able to run the power flow without encountering this error after synchronising and updating numpy, scipy and pandas. It seems to me that there is some incompatibility/was between these 4 modules(pp,numpy,scipy and pandas) which had led to this issue.

closed 2023-11-06 by AnkurArohi