e2nIEE/pandapower #2166: [bug] index error when running power flow calculation with added lines and switches

ced2205, 2023-11-08:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
net.line = 
        name                     std_type  from_bus  to_bus  length_km   
0   b1 to b2  NA2XS2Y 1x240 RM/25 6/10 kV         1       2   1.000000  \
1   b2 to b3  NA2XS2Y 1x240 RM/25 6/10 kV         2       3   0.247549   
2   b3 to b4  NA2XS2Y 1x240 RM/25 6/10 kV         3       4   0.204360   
3   b4 to b5  NA2XS2Y 1x240 RM/25 6/10 kV         4       5   0.369836   
4   b5 to b6  NA2XS2Y 1x240 RM/25 6/10 kV         5       6   0.403354   
5   b6 to b7  NA2XS2Y 1x240 RM/25 6/10 kV         6       7   0.229091   
6   b7 to b8  NA2XS2Y 1x240 RM/25 6/10 kV         7       8   0.815765   
7   b8 to b9  NA2XS2Y 1x240 RM/25 6/10 kV         8       9   0.361244   
8   b9 to b1  NA2XS2Y 1x240 RM/25 6/10 kV         9       1   0.900000   
9   b1 to b3  NA2XS2Y 1x240 RM/25 6/10 kV         1       3   1.000000   
10  b4 to b1  NA2XS2Y 1x240 RM/25 6/10 kV         4       1   0.800000   
11  b2 to b5  NA2XS2Y 1x240 RM/25 6/10 kV         2       5   0.633616   

    r_ohm_per_km  x_ohm_per_km  c_nf_per_km  g_us_per_km  max_i_ka   df   
0          0.122         0.105        456.0          0.0     0.416  1.0  \
1          0.122         0.105        456.0          0.0     0.416  1.0   
2          0.122         0.105        456.0          0.0     0.416  1.0   
3          0.122         0.105        456.0          0.0     0.416  1.0   
4          0.122         0.105        456.0          0.0     0.416  1.0   
5          0.122         0.105        456.0          0.0     0.416  1.0   
6          0.122         0.105        456.0          0.0     0.416  1.0   
7          0.122         0.105        456.0          0.0     0.416  1.0   
8          0.122         0.105        456.0          0.0     0.416  1.0   
9          0.122         0.105        456.0          0.0     0.416  1.0   
10         0.122         0.105        456.0          0.0     0.416  1.0   
11         0.122         0.105        456.0          0.0     0.416  1.0   

    parallel type  in_service  max_loading_percent  
0          1   cs        True                 80.0  
1          1   cs        True                 80.0  
2          1   cs        True                 80.0  
3          1   cs        True                 80.0  
4          1   cs        True                 80.0  
5          1   cs        True                 80.0  
6          1   cs        True                 80.0  
7          1   cs        True                 80.0  
8          1   cs        True                 80.0  
9          1   cs        True                 80.0  
10         1   cs        True                 80.0  
11         1   cs        True                 80.0

net.switch = 
    bus  element et type closed      name  z_ohm  in_ka in_service
0     1        0  l   CB   True   switch1    0.0    NaN       True
1     1        9  l   CB   True  switch10    0.0    NaN       True
2     2        1  l   CB  False   switch2    0.0    NaN       True
3     2       11  l   CB   True  switch12    0.0    NaN       True
4     3        2  l   CB  False   switch3    0.0    NaN       True
5     4        3  l   CB  False   switch4    0.0    NaN       True
6     4       10  l   CB   True  switch11    0.0    NaN       True
7     5        4  l   CB   True   switch5    0.0    NaN       True
8     6        5  l   CB  False   switch6    0.0    NaN       True
9     7        6  l   CB   True   switch7    0.0    NaN       True
[... 61 lines cut ...]
    raise KeyError(f"None of [{key}] are in the [{axis_name}]")
KeyError: "None of [Index([-2, -2, -1, -2, -1, -1, -2, -2, -1, -2, -2, -2], dtype='object')] are in the [index]"

```

### Expected Behavior

When I add new switches and lines to my network, this traceback appears. Before I added new lines and switches the timeseries powerflow calculation works. The initial Network where is no traceback in the timeseries powerflow is attached.
[mv_net_2040.json](https://github.com/e2nIEE/pandapower/files/13296482/mv_net_2040.json)

### Installed Versions

* python version:3.11 
* pandas version:2.0.0  
* networkx version:3.1  
* scipy version: 1.10.1 
* numpy version: 1.24.2 
* packaging version: 23.1 
* tqdm version:4.65.0  
* deepdiff version:6.3.0  
* Operating System name/version: Windows 11

### Label

- [X] Relevant labels are selected

rbolgaryn, 2023-11-23:
Hi @ced2205 ,

can you please run the function pp.diagnostic(net) and see if it outputs anything helpful?

ced2205, 2023-11-23:
Hi @rbolgaryn ,

i found the issue in results_branch.py in function _copy_switch_results_from_branches where open_switches=~net.switch.closed.values defined. this mask returns the index shown in the Error Log. I changed it to a list containing the indices of switches that are open/not closed and that worked for me.

vogt31337, 2024-11-21:
@ced2205 does this still occur? If not could you please close the issue?

vogt31337, 2025-03-10:
closed due to inactivity.

closed 2025-03-10 by vogt31337