e2nIEE/pandapower #2169: [Bug] Numba version interface has changed

DEUCE1957, 2023-11-14:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
In the current version of the PandaPower codebase, the version of Numba is checked in auxiliary.py by importing:

from numba._version import version_version as numba_version
```
However, in the latest versions of Numba (checked with Numba 0.58.1), the contents of numba._version are:
```
import json

version_json = '''
{
 "date": "2023-10-16T15:33:43+0200",
 "dirty": false,
 "error": null,
 "full-revisionid": "d4460feb8c91213e7b89f97b632d19e34a776cd3",
 "version": "0.58.1"
}
'''  # END VERSION_JSON

def get_versions():
    return json.loads(version_json)
```
So the new way to check the version would be to import:
```
from numba._version import get_versions
```
Then inside the function '_check_if_numba_is_installed':
```
def _check_if_numba_is_installed(numba:bool):
    try:
        # Get the numba version from the JSON description
        if version.parse(numba._version.get_versions()["version"]) < version.parse("0.2.5"):
            logger.warning('Warning: numba version too old -> Upgrade to a version > 0.25.\n' +
                           numba_warning_str)
            numba = False
    except:
        logger.warning(numba_warning_str)
        numba = False
    return numba
"""
The downside of this solution is that it may break again if numba._version changes again, perhaps append the original error message to the numba_warning_str (in my case the message was: "cannot import name 'version_version' from 'numba._version'").
```

### Issue Description and Traceback

If you comment out the try-except keywords in the "_check_if_numba_is_installed" function (auxiliary._check_if_numba_is_installed) then we get the following error:
Import Error: cannot import name 'version_version' from 'numba._version'

Otherwise a warning message is shown:
> numba cannot be imported and numba functions are disabled.
> Probably the execution is slow.
> Please install numba to gain a massive speedup.
> (or if you prefer slow execution, set the flag=False to avoid this warning)

### Expected Behavior

I expect no warning message nor error, since I have Numba installed and the version is greater than 0.2.5.

### Installed Versions

Python: 3.11.3
Numba: 0.58.1
Pandapower: 2.13.1

### Label

- [ ] Relevant labels are selected

rbolgaryn, 2023-11-23:
Hi @DEUCE1957 ,

thank you for the issue!

Can you please double-check if the issue is occurring on the develop branch?

BackMountainDevil, 2024-01-22:
python 3.10.11 with pandapower-2.13.1. numba warning version(0.57.0, 0.58.0, 0.58.1). numba-0.56.0 is fine. test code is [here](https://discourse.jupyter.org/t/numba-module-cannot-be-imported-properly/19353)

BDonnot, 2024-03-26:
Hello,

I am still getting this error:

> numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)

With the latest version of numba (0.59.1) and pandapower (2.13.1) on ubuntu 20.04.

I am pretty sure I got similar error messages for every version of numba (>= 0.58) and pandapower. Only way for me to silence these issues is either not to use numba (which is much slower) or downgrade to numba < 0.57 (which does not work on python >= 3.11)

vogt31337, 2025-03-10:
has changed in pandapower 3.0.0 and should not cause any issues anymore.

closed 2025-03-10 by vogt31337