e2nIEE/pandapower #2182: [bug] Inconsistent lines r_ohm_per_km and x_ohm_per_kmof case 14 with original definition

yamizi, 2023-11-29:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
net = pp.networks.case14()
pp.runpp(net)
error = pp.diagnostic(net)
print(error)
print(net.line["r_ohm_per_km"])
```

### Issue Description and Traceback

Hi,
It seems the r_ohm_per_km of case14 are very low and do not match at all the original definition.

the printed values:
[image]

The original definitions: https://labs.ece.uw.edu/pstca/pf14/ieee14cdf.txt

This leads the diagnostic tool to raise errors:
{'impedance_values_close_to_zero': [{'line': [7, 8, 9, 10, 11, 12, 13, 14]}]}
Indeed r_ohm_per_km of lines 7 to 14 are quite low.

My question is then why are the values different between pandapower and the original definition?
Matpower seems to have the correct values: https://matpower.org/docs/ref/matpower5.0/case14.html

Best,
Salah

### Expected Behavior

The expected values should match the definition in https://labs.ece.uw.edu/pstca/pf14/ieee14cdf.txt
Plus, I tried to load the matpower case: pp_net2 = pc.from_mpc('case14.m', f_hz=60) 
and it has bugs described in other tickets (not the core of this issue).

### Installed Versions

* python version: 3.11.5
* pandas version:  
* networkx version:  3.1
* scipy version:  1.11.3
* numpy version:  1.24.3
* packaging version:  23.1
* tqdm version:  4.65.0
* deepdiff version:  6.6.1
* Operating System name/version: Windows 11

### Label

- [X] Relevant labels are selected

BDonnot, 2024-07-11:
Hello,

From my experience here, the case14 of pandapower is correctly converted when "per unit" is applied. That is the values in the "per unit" files are correct:

> case["_ppc"]["internal"]["branch"][:, 2:5] 

which should match the information given in matpower file or the original file.

The really low values in pandapower are probably because the original ieee14 has no official voltage (in kV) and is expressed only in terms of per unit value (accessible in pandapower by looking at the internal `_ppc`). But in pandapower there are real voltages (bus["vn_kv"]).

However, I noticed (a while ago) that this was not the case for the ieee 118 as reported in issue #2131

vogt31337, 2024-11-21:
can this be closed?

BDonnot, 2024-11-22:
Hello,

Yes I think it has been resolved

heckstrahler, 2025-11-27:
Closed the model is wrong and will be recreated shortly.

closed 2025-11-27 by heckstrahler