e2nIEE/pandapower #2197: error while calculating 1ph sc capculations with fault resistance

trivedh13, 2024-01-03:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

hi sir, here i am sharing the error which i have got during the execution for calculating 1ph sc calculation
please suggest me to address this issue
KeyError                                  Traceback (most recent call last)
File ~\anaconda3\lib\site-packages\pandas\core\indexes\base.py:3802, in Index.get_loc(self, key, method, tolerance)
   3801 try:
-> 3802     return self._engine.get_loc(casted_key)
   3803 except KeyError as err:

File ~\anaconda3\lib\site-packages\pandas\_libs\index.pyx:138, in pandas._libs.index.IndexEngine.get_loc()

File ~\anaconda3\lib\site-packages\pandas\_libs\index.pyx:165, in pandas._libs.index.IndexEngine.get_loc()

File pandas\_libs\hashtable_class_helper.pxi:5745, in pandas._libs.hashtable.PyObjectHashTable.get_item()

File pandas\_libs\hashtable_class_helper.pxi:5753, in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'x0x_min'

The above exception was the direct cause of the following exception:

KeyError                                  Traceback (most recent call last)
Cell In[6], line 53
     51 net.ext_grid['rx_min'] = 0.1
     52 net.line["endtemp_degree"]=20
---> 53 sc.calc_sc(net, bus=1, fault="1ph", case="min", ip=True, ith=True, lv_tol_percent=10, topology='radial', r_fault_ohm=20, x_fault_ohm=0, branch_results= True, return_all_currents= True, inverse_y=True)
     55 print(net.res_bus_sc)
     56 print(net.res_line_sc)

File ~\anaconda3\lib\site-packages\pandapower\shortcircuit\calc_sc.py:161, in calc_sc(net, bus, fault, case, lv_tol_percent, topology, ip, ith, tk_s, kappa_method, r_fault_ohm, x_fault_ohm, branch_results, check_connectivity, return_all_currents, inverse_y, use_pre_fault_voltage)
    159     _calc_sc(net, bus)
    160 elif fault == "1ph":
--> 161     _calc_sc_1ph(net, bus)
    162 else:
    163     raise ValueError("Invalid fault %s" % fault)

File ~\anaconda3\lib\site-packages\pandapower\shortcircuit\calc_sc.py:239, in _calc_sc_1ph(net, bus)
    236 _calc_ybus(ppci)
    238 # zero seq bus impedance
--> 239 ppc_0, ppci_0 = _pd2ppc_zero(net, ppc['branch'][:, K_ST])
    240 _calc_ybus(ppci_0)
    242 if net["_options"]["inverse_y"]:

File ~\anaconda3\lib\site-packages\pandapower\pd2ppc_zero.py:47, in _pd2ppc_zero(net, k_st, sequence)
     45 _build_tcsc_ppc(net, ppc, "sc")  # needed for shape reasons
     46 _add_gen_sc_impedance_zero(net, ppc)
---> 47 _add_ext_grid_sc_impedance_zero(net, ppc)
     48 _build_branch_ppc_zero(net, ppc, k_st)
     50 # adds auxilary buses for open switches at branches

File ~\anaconda3\lib\site-packages\pandapower\pd2ppc_zero.py:426, in _add_ext_grid_sc_impedance_zero(net, ppc)
    424     r0_grid = net.ext_grid["r0x0_%s" % case].values * x0_grid
    425 elif case == "min":
--> 426     x0_grid = net.ext_grid["x0x_%s" % case].values * x_grid
    427     r0_grid = net.ext_grid["r0x0_%s" % case].values * x0_grid
    428 y0_grid = 1 / (r0_grid + x0_grid*1j)

File ~\anaconda3\lib\site-packages\pandas\core\frame.py:3807, in DataFrame.__getitem__(self, key)
   3805 if self.columns.nlevels > 1:
   3806     return self._getitem_multilevel(key)
-> 3807 indexer = self.columns.get_loc(key)
   3808 if is_integer(indexer):
   3809     indexer = [indexer]

File ~\anaconda3\lib\site-packages\pandas\core\indexes\base.py:3804, in Index.get_loc(self, key, method, tolerance)
   3802     return self._engine.get_loc(casted_key)
   3803 except KeyError as err:
-> 3804     raise KeyError(key) from err
   3805 except TypeError:
   3806     # If we have a listlike key, _check_indexing_error will raise
   3807     #  InvalidIndexError. Otherwise we fall through and re-raise
   3808     #  the TypeError.
   3809     self._check_indexing_error(key)

KeyError: 'x0x_min'
​

### Label

- [ ] Relevant labels are selected

rbolgaryn, 2024-03-04:
Hi @trivedh13 ,

this means that the optional value "x0x_min" is missing from the element table net.ext_grid

pandapower requires additional (optional) parameters to be added in order to execute a short-circuit calculation

rbolgaryn, 2024-03-04:
Please refer to the documentation to see what additional parameters you need to provide for each element

closed 2024-03-04 by rbolgaryn