e2nIEE/pandapower #2218: [3ph powerflow] Power Factor of asymmetric loads in the IEEE European LV Test Feeder

Karelix, 2024-02-07:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hello, 

I am examining the deafault scenario of the European LV Test Feeder: 
```python
net = pp.networks.ieee_european_lv_asymmetric(scenario='on_peak_566')
```
By examining the asymmetric loads of this scenario, I noticed that the Power Factor of loads connected in Phase B is not equal to 0.95, as dictated by the [data provided](https://cmte.ieee.org/pes-testfeeders/wp-content/uploads/sites/167/2017/08/European_LV_Test_Feeder_v2.zip) by IEEE. 

More specifically, in the Loads.csv file that exists in the European_LV_CSV directory in the data folder, the PF column is 0.95 for all loads, however, the pandapower's asymmetric loads connected of Phase B for the 'on_peak_566' yield a Power Factor of 1.

Also, in scenario 1440, I noticed that loads connected in Phases B and C are 1000 times smaller than the ones dictated by the data and calculation method from IEEE.

Here is my code for scenario 566:

```python
import pandas as pd
import pandapower as pp
import numpy as np

net = pp.networks.ieee_european_lv_asymmetric(scenario="on_peak_566")

loads = net.asymmetric_load.copy()

loads["phase_a"] = loads["p_a_mw"] + 1j * loads["q_a_mvar"]
loads["phase_b"] = loads["p_b_mw"] + 1j * loads["q_b_mvar"]
loads["phase_c"] = loads["p_c_mw"] + 1j * loads["q_c_mvar"]

# Monophasic loads, so I only need the non-zero phase.
phases = loads[["phase_a", "phase_b", "phase_c"]].to_numpy()
loads["non_zero_phase"] = phases[phases != 0]
loads["power_factor"] = np.cos(np.angle(loads["non_zero_phase"]))
loads["P"] = np.real(loads["non_zero_phase"])
loads["Q"] = np.imag(loads["non_zero_phase"])
print(loads[["name", "bus", "P", "Q", "power_factor"]])
```
Here is the code's output:
```output
      name  bus         P             Q  power_factor
0    LOAD1   34  0.000574  1.886647e-04          0.95
1    LOAD2   47  0.000044  1.446210e-08          1.00
2    LOAD3   70  0.000114  3.746999e-05          0.95
3    LOAD4   73  0.000231  7.592603e-05          0.95
4    LOAD5   74  0.000035  1.150394e-05          0.95
5    LOAD6   83  0.000395  1.298302e-07          1.00
6    LOAD7  178  0.000055  1.807763e-08          1.00
7    LOAD8  208  0.002836  9.321481e-07          1.00
8    LOAD9  225  0.000131  4.305762e-05          0.95
9   LOAD10  248  0.002798  9.196581e-07          1.00
10  LOAD11  249  0.000049  1.610552e-08          1.00
11  LOAD12  264  0.000286  9.400365e-08          1.00
12  LOAD13  276  0.002211  7.267205e-07          1.00
13  LOAD14  289  0.000210  6.902366e-05          0.95
14  LOAD15  314  0.002331  7.661627e-07          1.00
15  LOAD16  320  0.000510  1.676289e-07          1.00
16  LOAD17  327  0.000048  1.577684e-08          1.00
17  LOAD18  337  0.000411  1.350892e-07          1.00
18  LOAD19  342  0.000457  1.502086e-07          1.00
19  LOAD20  349  0.000051  1.676289e-05          0.95
20  LOAD21  387  0.000045  1.479078e-05          0.95
21  LOAD22  388  0.000054  1.774894e-05          0.95
22  LOAD23  406  0.000236  7.756945e-08          1.00
23  LOAD24  458  0.000075  2.465131e-08          1.00
24  LOAD25  502  0.000234  7.691208e-05          0.95
25  LOAD26  522  0.012659  4.160812e-06          1.00
26  LOAD27  539  0.000155  5.094603e-08          1.00
27  LOAD28  556  0.000028  9.203155e-09          1.00
28  LOAD29  562  0.010471  3.441651e-03          0.95
29  LOAD30  563  0.000053  1.742026e-05          0.95
30  LOAD31  611  0.002283  7.503858e-04          0.95
31  LOAD32  614  0.000606  1.991826e-07          1.00
32  LOAD33  619  0.000303  9.959128e-08          1.00
33  LOAD34  629  0.000058  1.906368e-05          0.95
34  LOAD35  639  0.005570  1.830770e-06          1.00
35  LOAD36  676  0.000156  5.127472e-08          1.00
36  LOAD37  682  0.001063  3.493912e-07          1.00
37  LOAD38  688  0.000392  1.288442e-07          1.00
38  LOAD39  701  0.000048  1.577684e-08          1.00
39  LOAD40  702  0.000427  1.403481e-07          1.00
40  LOAD41  755  0.000065  2.136447e-08          1.00
41  LOAD42  778  0.000366  1.202984e-07          1.00
42  LOAD43  780  0.000050  1.643420e-08          1.00
43  LOAD44  785  0.001677  5.512032e-07          1.00
44  LOAD45  813  0.000173  5.686235e-08          1.00
45  LOAD46  817  0.000049  1.610552e-05          0.95
46  LOAD47  835  0.000045  1.479078e-08          1.00
47  LOAD48  860  0.002141  7.037127e-04          0.95
48  LOAD49  861  0.000054  1.774894e-05          0.95
49  LOAD50  886  0.000310  1.018921e-07          1.00
50  LOAD51  896  0.000053  1.742026e-05          0.95
51  LOAD52  898  0.000295  9.696181e-05          0.95
52  LOAD53  899  0.003087  1.014648e-06          1.00
53  LOAD54  900  0.000245  8.052761e-05          0.95
54  LOAD55  906  0.000055  1.807763e-05          0.95
```
I notice that for scenario 566, the imaginary part of the load (Q), only for loads connected in phase B, is 1000 smaller than expected.

I think that I understand the IEEE's calculation method correctly, but I might be wrong, since it is not so straightforward.

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-11-21:
@Karelix have you resolved this issue?

Karelix, 2024-11-21:
Hello, this was a question regarding the power factors of the loads in the European LV Test Feeder. Since the documentation of the feeder dictates that all power factors of all loads should be 0.95, I thought that the default configuration included in the pandapower library was wrong. The issue is that changing the loads to what I think it is the correct values does not give the correct results. I have tested this with an asymmetric PFA that I have developed. When I changed the values to the seemingly "wrong" ones I get the correct results. Do you have any further info regarding the power factors and why they should not be 0.95 in certain loads?