e2nIEE/pandapower #2257: [bug] PowerFactory export validation fails when number of switches is different between PF and PP

Heiner92, 2024-04-05:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
I used the PF exporter on a meshed low voltage grid.
```

### Issue Description and Traceback

The PowerFactory export GUI is a handy tool. Unfortunately there is a bug in the validation part of the exporter.  
I used the following settings: 
[image]
The following output get displayed in the PF console:
```Validating import ...
set 5 lines out of service
set 0 impedances out of service
running pandapower loadflow
pandapower net converged: True
Error while verifying net: Unalignable boolean Series provided as indexer (index of the boolean Series and of the indexed object do not match).
Traceback (most recent call last):
File "C:\Users\Heiner. Frueh\Documents\Test\Lib\site-packages\pandapower\converter\powerfactory\main_pf.py", line 195, in calc
run_verify(net)
File "C:\Users\Heiner. Frueh\Documents\Test\Lib\site-packages\pandapower\converter\powerfactory\main_pf.py", line 134, in run_verify
all_diffs = validate_pf_conversion(net, ** load_flow_params)

File "C:\Users\Heiner. Frueh\Documents\Test\Lib\site-packages\pandapower\converter\powerfactory\validate.py", line 269, in validate_pf_conversion
wrong_switches = net.res_switch. loc[

File "C:\Python311\Lib\site-packages\pandas\core\indexing.py", line 1153, in _getitem_
return self ._ getitem_axis(maybe_callable, axis=axis)

File "C:\Python311\Lib\site-packages\pandas\core\indexing.py", line 1375, in _getitem_axis
return self ._ getbool_axis(key, axis=axis)

File "C:\Python311\Lib\site-packages\pandas\core\indexing.py", line 1171, in _getbool_axis
key = check_bool_indexer(labels, key)

File "C:\Python311\Lib\site-packages\pandas\core\indexing.py", line 2575, in check_bool_indexer
raise IndexingError(
pandas.errors. IndexingError: Unalignable boolean Series provided as indexer (index of the boolean Series and of the indexed object do not match).
Python-Skript 'convert2pandapower' erfolgreich ausgeführt
```

For some unknown reason the number of switches is different between PF and PP. The bug is caused by lines 268-271 in converter/powerfactory/validate.py:
```
pf_closed = pf_results['pf_switch_status']
wrong_switches = net.res_switch. loc[
    pf_closed != net.switch. loc[pf_closed.index, 'closed']
].index.values if 'res switch' in net.keys() else []
```
I fixed it by changing it into:
```
pf_closed = pf_results['pf_switch_status']
wrong_switches = pf_closed. loc[
    pf_closed != net.switch. loc[pf_closed.index, 'closed']
].index.values if 'res switch' in net.keys() else []
```

### Expected Behavior

I can see the rest of the validation output.

### Installed Versions

DlgSILENT PowerFactory 2023
python: 3.11
pandapower: 2.14.6

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/FRHHEI~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/FRHHEI~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

os | Windows
-- | --
OS-release | 10
Version | 10.0.19045
machine | AMD64
processor | Intel64 Family 6 Model 85 Stepping 7, GenuineIntel
byteorder | little
LC_ALL | None
LANG | None
LOCALE | de_DE. cp1252
  |  
pandas | 02.01.2002
numpy | 1.25.2
pytz | 2023.3. post1
dateutil | 02.08.2002
setuptools | 65.5.0
pip | 23.02.2001
Cython | 3.0.5
'pytest | 07.04.2003
hypothesis | 6.88.3
sphinx | 07.02.2006
blosc | 01.11.2001
feather | 0.4.1
xlsxwriter | 03.01.2009
1xml. etree | 04.09.2003
htm151ib | 01. Jan
pymysq1 | None
psycopg2 | None
jinja2 | 03.01.2002
IPython | 8.17.2
pandas_datareader | None
bs4 | 04.12.2002
bottleneck | 01.03.2007
dataframe-api-compat | None
fastparquet | 01.10.2023
fsspec | 2023.10.0
gcsfs | None
'matplotlib | 03.08.2001
numba | 0.58.1
numexpr | 02.08.2007
odfpy | None
openpyx1 | 03.01.2002
pandas_gbq | None
pyarrow | 14.0.0
pyreadstat | None
pyx1sb | None
s3fs | None
scipy | 01.11.2003
sqlalchemy | 2.0.23
'tables | 03.09.2001
tabulate | 0.9.0
xarray | 01.10.2023
x1rd | 2.0.1
zstandard | 0.22.0
tzdata | 2023.3
qtpy | 02.04.2001
pyqt5 | None

</body>

</html>

### Label

- [ ] Relevant labels are selected

rbolgaryn, 2024-04-12:
Thank you! I think it could be due to the exporter replacing low-impedance lines by switches, creating new switches. I will fix it.

Heiner92, 2024-04-12:
Thank you very much! As far as I know this bug was also present in the previous version, where the option to replace low-impedance lines with switches was not yet present (at least not in the GUI). 

Could you please also add an option to specify the accuracy of the validation output or the units (e.g. kVA instead of MVA), or make it dependet on the lowest nominal voltage? Our use case includes LV-grids, which are tricky to model in PF because of the many scaling factors that may or may not be present. Hence we would like to be able to see any differences in loadflow results after the export. Currently the validation is displayed with 0.1 MVA accuracy. I could also make a PR sometime next week for you to decide.  

In case you need a pfd grid for testing, please get in contact with me :)