e2nIEE/pandapower #2271: [bug] YOUR TITLE

JasmineGarland, 2024-04-20:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
n/a
```

### Issue Description and Traceback

There is a disconnect with  network convergence and using the pandapower toolbox. I run a time series ACOPF with continue_on_divergence=False and it shows that it runs 100% of the time steps/no errors. However, when I try to use toolbox metrics such as, pandapower.overloaded_lines(net, max_load=100) - I get the following message: 
UserWarning("The last loadflow terminated erratically, results are invalid!"). Which I see in the code (result_info.py) means that net.converaged did not pass....? 

Is there an explanation for this behavior?

Thanks!

### Expected Behavior

I would expect that either time series module would show an error or the toolbox features would work.

### Installed Versions

* python version: * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:
python: 3.8.18, pandas : 2.0.3, numpy : 1.23.0, numba  : 0.56.4, 

### Label

- [X] Relevant labels are selected

vogt31337, 2024-06-13:
A simple example would be nice, and could add which pandapower version you used?

vogt31337, 2024-11-21:
@JasmineGarland ?

closed 2025-03-10 by vogt31337