e2nIEE/pandapower #2284: Code doesn't execute.

i5sanchez, 2024-05-11:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
# -*- coding: utf-8 -*-
"""
Created on Thu May  9 15:30:17 2024
@author: Usuario
"""
import pandapower as pp
import pandas as pd
import os
os.environ["NUMBA_DISABLE_JIT"] = "1"

Vdc = 12

# Crear una red vacía
net = pp.create_empty_network()

#print(pp.available_std_types(net, "line"))

# Código para corregir los valores NaN en net.ext_grid
net.ext_grid['name'] = net.ext_grid['name'].fillna('bus_1')
net.ext_grid['bus'] = net.ext_grid['bus'].fillna(1)
net.ext_grid['bus'] = net.ext_grid['bus'].astype(int)

# Crear el bus 1 con coordenadas geográficas
bus1 = pp.create_bus(net, vn_kv=0.012, name='bus_1', geodata=(-65.4644, -18.169))

# Crear el bus 2 con coordenadas geográficas
bus2 = pp.create_bus(net, vn_kv=0.012, name='bus_2', geodata=(-65.4634, -18.1704))

# Crear el bus 3 con coordenadas geográficas
bus3 = pp.create_bus(net, vn_kv=0.012, name='bus_3', geodata=(-65.465, -18.1699))

net.bus_geodata['coords'] = net.bus_geodata.apply(lambda row: (row['x'], row['y']), axis=1)

print(net.bus_geodata.isnull().sum())

# Agregar almacenamiento de energía a cada bus
storage_capacity = 712  # Capacidad de almacenamiento en Wh
max_p_charge = 160  # Potencia máxima de carga en W
max_p_discharge = 160  # Potencia máxima de descarga en W
efficiency = 0.95  # Eficiencia de carga/descarga

for bus in [bus1, bus2, bus3]:
    pp.create_storage(net, bus=bus, p_mw=0, max_e_mwh=storage_capacity / 1000, max_p_mw=max_p_charge, min_p_mw=-max_p_discharge, efficiency=efficiency)

# Agregar la columna 'state_of_charge_mwh' al DataFrame net.storage e inicializarla con ceros
net.storage['state_of_charge_mwh'] = 0.0

# Ruta del archivo CSV a importar
ruta_archivo = "C:\\Users\\Usuario\\.spyder-py3\\data_proysolar\\df_4viviendas_30.97.csv"

# Cargar el archivo CSV en un DataFrame
df = pd.read_csv(ruta_archivo)

dc_line_std_type = {
    "r_ohm_per_km": 1.95,  # Resistencia de la línea en ohmios/km
[... 169 lines cut ...]
  File ~\anaconda3\envs\ramp\lib\site-packages\pandapower\auxiliary.py:798 in _select_is_elements_numba
    set_elements_oos(element_df["bus"].values, element_df["in_service"].values,

  File ~\anaconda3\envs\ramp\lib\site-packages\numba\core\dispatcher.py:468 in _compile_for_args
    error_rewrite(e, 'typing')

  File ~\anaconda3\envs\ramp\lib\site-packages\numba\core\dispatcher.py:409 in error_rewrite
    raise e.with_traceback(None)

TypingError: non-precise type array(pyobject, 1d, C)
During: typing of argument at C:\Users\Usuario\anaconda3\envs\ramp\lib\site-packages\pandapower\auxiliary.py (756)

El sistema no puede encontrar la ruta especificada.

### Expected Behavior

It should give the results and power fluxes in the grid.

### Installed Versions

* python version: 3.10.14 * pandas version: 2.2.2 * networkx version: 3.3  * scipy version: 1.13.0  * numpy version: 1.24.4  * packaging version: 23.2 * tqdm version: 4.66.2 * deepdiff version: 7.0.1 * Operating System name/version:  Microsoft Windows 11 Home /  10.0.22631 N/D Compilación 22631

### Label

- [X] Relevant labels are selected

vogt31337, 2024-06-13:
which pandapower version did you use?

vogt31337, 2024-11-21:
@i5sanchez still relevant?

vogt31337, 2025-03-10:
I cannot check your code since data is missing. Therefore I cannot help you, maybe a suggestion try your code in steps and have a look when it does not work anymore.

closed 2025-03-10 by vogt31337