e2nIEE/pandapower #2286: [Question] Plotting a topology structure

bacardischmal, 2024-05-19:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Is there a way to draw the created network as a kind of topology? I only found the Simple Plotting options, but i would like to plott a topology structure without the geodata strucure.

is this possible i'm looking for a schema like this
https://pandapower.readthedocs.io/en/v2.13.1/_images/example_network_simple.png
or
https://pandapower.readthedocs.io/en/v2.0.0/_images/kerber_landnetz_kabel.png

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-06-13:
Sadly no. 
It is really complex to create these plots only based on topology information. You can generate generic coordinates via the topology package and use force directed graph for example. But It won't look anything like your example.
Feel free to build something and create a PR.

vogt31337, 2024-11-21:
@bacardischmal still relevant?

bacardischmal, 2024-11-30:
> @bacardischmal still relevant?
yes its still relevant for me but I cant create it by myself.

vogt31337, 2025-03-10:
closed since such a feature is nearly impossible to implement, in an automated fashion.

closed 2025-03-10 by vogt31337