e2nIEE/pandapower #2287: DC Line parameters

joeygysf, 2024-05-21:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

I would like to compare the performance of both an MVAC and an MVDC link between two subnetworks.
I am facing trouble specifying the parameters required for the command 'pp.create_dcline ' such as 'vm_from_pu', 'vm_to_pu',
and 'p_mw'. This command is different from that of the AC line that takes as input the R,L,C of the line.

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-06-13:
Maybe I can help a bit, a DC-Line is modeled by using two generators. Therefore you have to provide all these parameters. 
The results in an AC-Line are provided by the powerflow calculation itself. So DC you need to provide what is happening. AC you get told what happened.
Maybe a solution is to look at the voltage angles on the two dc-line busses and compare them. If they are "near" then the system is balanced. If not you need to readjust the parameters. Try searching for strategies on how to operate dc-lines. This voltage angle based approach is a quite common one.

vogt31337, 2024-11-21:
@joeygysf still relevant?