e2nIEE/pandapower #2288: How to adjust steps of shunt devices in time series simulation

mchirsa5, 2024-05-22:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

I'm running the time series simulation for a distribution system. In the simulation, I will have hundreds of transformers and shunt capacitors. I want to adjust the tap of the transformers and the steps of the c-bank based on the voltage magnitudes of the connected bus. Currently, I can change the taps of transformers in the network using "trafocontroller". However, I don't know how to adjust the steps of the c-bank based on the voltage magnitude of the connected bus. Could you advise if there is any function to do that kind of adjustment?

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-06-13:
Can't you just set the step of the shunt? If this is the case, you could write a shunt controller similar to the trafo controller and just add it. Maybe write one and send a Pull Request.

pawellytaev, 2024-08-29:
Hi @mchirsa5, we just merged two new controllers that might solve your issue: A DiscreteShuntController https://github.com/e2nIEE/pandapower/blob/develop/pandapower/control/controller/shunt_control.py for changing the discrete steps of the shunt depending on voltage settings (similar to DiscreteTapControl for trafos) 
and BinarySearchControl https://github.com/e2nIEE/pandapower/blob/develop/pandapower/control/controller/station_control.py that you can use if you're relaxing the steps of the shunts

you can find uses for both here: https://github.com/e2nIEE/pandapower/blob/develop/pandapower/test/control/test_shunt_control.py

HafizRiaz, 2024-09-07:
> ### Feature Checklist
> 
>     * [x]  Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
> 
>     * [x]  Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)
> 
>     * [x]  Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
> 
> 
> ### Issue
> 
> I'm running the time series simulation for a distribution system. In the simulation, I will have hundreds of transformers and shunt capacitors. I want to adjust the tap of the transformers and the steps of the c-bank based on the voltage magnitudes of the connected bus. Currently, I can change the taps of transformers in the network using "trafocontroller". However, I don't know how to adjust the steps of the c-bank based on the voltage magnitude of the connected bus. Could you advise if there is any function to do that kind of adjustment?
> ### Label
> 
>     * [x]  Relevant labels are selected
> 
> 
> ### Tasks
> Beta [Give feedback](https://github.com/community/community/discussions/39106)
> No tasks being tracked yet.
> 
>     1. [ ] 
>                      
>                    
>        
>                    
>                      
>                        
>        
>                      
>                        
>                    
>                  
>                  
>                    
>          
>              
>            
>        
>        Options
>        
>        
>        
>          
>            
>                        
>          
>            
>                * Convert to issue
>                * Toggle completion
>                * Rename
>                * Remove
> 
>     2.

HafizRiaz, 2024-09-07:
I ma trying to add capacitor to my existing code  that is currently dealing with continuous devices like inverter and SVC, using DDPG algo, i am wondering if there is a discrete controller for shunt capacitor that dont allow taps, just switches On or OFF when voltage falls below lower limit e.g 0.95 p.u, plz help me regarding this

HafizRiaz, 2024-09-07:
> Hi @mchirsa5, we just merged two new controllers that might solve your issue: A DiscreteShuntController [...]

i try to use these shunt controllers, but it raised no module found error as attached
[image]

pawellytaev, 2024-09-10:
if you set up the shunt with max_step=1, and use the discrete shunt controller, it should be able to turn it on or off. otherwise you can always build your own controller with this functionality, here's a tutorial that explains it: https://github.com/e2nIEE/pandapower/blob/develop/tutorials/building_a_controller.ipynb

for the not-converging power flow, we'll need more information on the grid. does it work with newton raphson? does the DC power flow calculate? maybe you can start with it. Since it's a separate topic from this issue here, please open another issue or discussion

HafizRiaz, 2024-09-12:
> if you set up the shunt with max_step=1, and use the discrete shunt controller, it should be able to turn it on or off. otherwise you can always build your own controller with this functionality, [...]

Thank you for your suggestion. I was wondering, what if, i want to apply more than one capacitor along with discrete shunt controller, in that case, "shunt_index=0, bus_index=1," will be changed by adding commas like this? "shunt_index=0,1, bus_index=1,2" or i have to try rewrite the command for other controller

vogt31337, 2024-11-21:
@HafizRiaz is your problem solved?

HafizRiaz, 2024-11-22:
Yes it is solved. Thanks for your time and consideration