e2nIEE/pandapower #2290: [feature] Expose sensitivity measures such as PTDF

leuchtum, 2024-05-23:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [X] Adding new functionality to pandapower

- [X] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

Sensitivity measures such as the power transfer distribution factors (PTDF) are part of modern power network analytics. They are commonly used in market analytics, congestion management, stability analysis and more. Currently the PTDFs are calculated by `pypower.makePTDF` and therefore `_pd2ppc` needs to be used by the user. This is error prone, as one needs to deal with indices and more.

### Feature Description

I suggest to expose the calculation of sensitivity measures directly, something like `pp.ptdf(net)`. There has already been interest in this feature, as #847, #1419, #1572 and #1622 have shown.

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-06-13:
Sounds like an interesting topic, could you provide more information on this topic? I have until now never used ptdf and maybe could you estimate how complex implementing this behaviour is? Or trace a bit more what needs to be changed?

vogt31337, 2024-07-26:
@leuchtum, could you provide more insight?

leuchtum, 2024-07-26:
@vogt31337 Sorry for not answering a month, I've been involved in other projects away from our power-network-modeling-world. However, I can foresee that I will be more active on the topic (especially PTDF's) again as from September. Would it be ok if we leave the issue open until then and I'll get back to you when I'm deeper into the topic again?

vogt31337, 2024-07-26:
No Problem. I find this an interesting topic and used a brute force approach for calculating sensitivities. That's why i asked. See you in September.

leuchtum, 2024-08-28:
Also mentioned in https://github.com/e2nIEE/pandapower/issues/833#issuecomment-1078714536

AnkurArohi, 2024-08-28:
This is a good feature to have, currently the function needs _pd2ppc with branches and nodes, this is very problematic because one has to match the indices , as already mentioned above, Now when one gets the results again one needs to interpret back to the actual pandapwoer network elements.

Basically, what is suggested here is to be able to provide instead of pd2ppc branches and nodes, simply pandapower lines, trafos, trafo3w, switches etc and busses, now the conversion of these to pypower pd2ppc should be mart of makePTDF func, this is simple in pandapower as there are already such functions to map the pandapower to pypower pd2ppc

leuchtum, 2024-08-28:
@AnkurArohi exactly. 

However, you could consider thinking a little bigger and creating a kind of submodule for sensitivity analyses in general, because PTDF matrices, for example, are only one part of the possible sensitivity analyses. 

Even with the PTDF matrices themselves, there are various possibilities. The current function `makePTDF` uses a DC approximation, but there are many approaches to developing a PTDF matrix around an AC solution. Both have advantages and disadvantages. In this context, it is worth to read the paper by Baldick et al (Here they write about incremental PTDFs (=AC-PTDFs) and DC PTDFs): https://ieeexplore.ieee.org/document/1489123

@AnkurArohi what is your way of calculating a PTDF Matrix right now? Can you give an example? I want to start to collect a bit of examples to see where we are going with this.

AnkurArohi, 2024-08-28:
@leuchtum 
Nein, The thing is you actually do not need increments to calculate "better" AC-PTDF matrices. For AC sensitivity analysis we should be able to extract the Jacobian Matrix (just before the power flow in pypower).

Here we are concerned currenlty only with the DC Approximated PTDF.

My way of calculating the PTDF matrix is to provide the function with pd2ppc branches and nodes, then interpret (map) the results back to pandapower elements.

To summarize we need
1. make_PTDF func which can accept pandapower elements and returns pandapwoer elements sens
2. AC sens analysis should be possible through the use of jacobian matrix

vogt31337, 2024-11-21:
@AnkurArohi @leuchtum feel free to send a PR.

vogt31337, 2026-11-11:
Merged with #2822

closed 2026-11-11 by vogt31337