e2nIEE/pandapower #2302: Chinese documents

lambxk, 2024-06-09:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hello, do you have Chinese help documents? I am a Chinese, and I have some problems when using this tool. My English is not very good, so I would like to ask if there are any Chinese documents for support

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-06-13:
Hi, have you tried using translation services? What exactly is your question? Maybe I can help you, but I don't speak chinese and as far as I know there is no documentation in chinese available...

rbolgaryn, 2024-06-27:
Hi @lambxk ,

unfortunately, we only have documentation in English and we do not plan to include other languages in the future,