e2nIEE/pandapower #2305: Advice on pre-fault voltage factor for short circuit calculation

mrsilvabr, 2024-06-11:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hi there
I am using the software for short circuit levels investigations. I came upon the following situation:

circuit(violet2)
bus 0 - 69 kV
bus 1 - 69 kV
external grid (69 kV, 1000 MVA, rx=22)
external grid----->bus0(source_1)-->line(CB27_LT)-->bus1(T2PRIM).

It is a very simple circuit, downsized from fig. 4-1 - IEEE Std 551-2006 - Violet Book
The short circuit level at bus1(T2PRIM) evaluated by hand is 6797 A. Essentially the same as resulting from other packages.
The result from Panda Power is 7135 A, which I found to be 1.05 x 6797. Now I know that, given the 69 kV system, a factor of 1.1 should be applied as per the IEC 69909 recommendation. That should give me 7476 A in Panda Power.

Would someone comment on this?

Thanks

### Label

- [X] Relevant labels are selected

vogt31337, 2024-07-26:
I don't do short ciruit investigations. So is this to be considered a bug? If so could you write a test?

mrsilvabr, 2024-07-27:
Thank you for the interest. As far as I see, the calculation by hand is the best test I can do.
I am not knowledgeable enough to check the internal routines of the software.
Regards

gbanerjeefraunhofer, 2024-07-29:
Dear @mrsilvabr ,
please provide your hand calculation and your pandapower calculation code so we can evaluate the difference.
In pandapower, C factor is considered as 1.1 based on IEC60909 code for the buses over 1 kV voltage level. 
Please follow the attached link and check your input parameter accordingly.
https://github.com/e2nIEE/pandapower/blob/ed0e8e534193927b6bfe0d5a31fd8014e7dc0cca/pandapower/build_bus.py#L716

Please recalculate, check and compare the result and get back to us if you have still mismatch result.

mrsilvabr, 2024-08-20:
gbanerjeefraunhofer, thank you for the answer.
I have been involved in a new project, with my hands an head full. I will do as you instructed me but it is going to take a few weeks.
I will get back as soon as I can.

gbanerjeefraunhofer, 2024-11-26:
Dear @mrsilvabr ,

We have been waiting for your response on this issue since 20th Aug.
We would be glad to know if you have any further information or progress on it.

Thanks and regards,
Gourab Banerjee

mrsilvabr, 2024-12-05:
Hi dear Gourab.
 I hope everybody in this community is fine and wish a great new year for all!
I just checked my calculations. I have attached 2 files (Panda Power simulation and by-hand calculation using electrical system data from Violet Book - fig. 4.15). Please try to download them from the links below. I have not been able to attach files to this message.
https://www.dropbox.com/scl/fi/dxegvuif81fn5xshbv5sb/Exemplo-VioletBook4_15.ipynb?rlkey=2ix807fanr712imaz4cmycqkj&dl=0
https://www.dropbox.com/scl/fi/tb4cu4q14ez1lcjl218fr/Exemplo_VioletBook_1.ipynb?rlkey=dqkhrojda6iqc0c9y12og8ogc&dl=0.

My conclusions were:
The short circuit current obtained in the "by hand" exercise is essentially the same as Panda Power delivers (6.8 kA). In this exercise however, no factor has been applied to the source voltage. I am forced to conclude that Panda Power is giving me a smaller result for the similar conditions. If I were to apply the 1.1 factor to my calculations I would be 10% above Panda Power results.

It is worth to mention that the bus 0 impedance shown in Panda Power, in the matrix violet1.res_bus_sc, is 10% higher than the impedance obtained in this exercise. Check the matrix and the variable "ZUTIL"  in the file.

Thank you

mrsilvabr, 2024-12-05:
Hi dear Gourab.
 I hope everybody in this community is fine and wish a great new year for all!
I just checked my calculations. I have attached 2 files (Panda Power simulation and by-hand calculation using electrical system data from Violet Book - fig. 4.15). 

My conclusions were:
The short circuit current obtained in the "by hand" exercise is essentially the same as Panda Power delivers (6.8 kA). In this exercise however, no factor has been applied to the source voltage. I am forced to conclude that Panda Power is giving me a smaller result for the similar conditions. If I were to apply the 1.1 factor to my calculations I would be 10% above Panda Power results.

It is worth to mention that the bus 0 impedance shown in Panda Power, in the matrix violet1.res_bus_sc, is 10% higher than the impedance obtained in this exercise. Check the matrix and the variable "ZUTIL"  in the file.

Thank you￼￼

mrsilvabr, 2024-12-15:
Hi there
I revisited this subject, performed my calculations again and concluded that the differences in results I was getting to were related to the application of the "C factor" mandated by the 60909 Standard. I was able to find very comparable results once I followed the standard DIN/IEC EN 60909.
The paper by Mr Ger Belzer - SHORT-CIRCUIT CALCULATION WITH FULLSIZE CONVERTERS ACCORDING
TO IEC 60909 was of great help.
So I am closing the issue.
Than you

closed 2024-12-15 by mrsilvabr

gbanerjeefraunhofer, 2024-12-16:
@mrsilvabr ,

It is great news that you found the solution and can use the pandapower short-circuit calculation successfully.
All the best for the future use of pandapower.