e2nIEE/pandapower #2322: Pandapower is missing generator from 9-bus ieee case

Paulnkk, 2024-06-28:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hey,

I currently want to use pandapower in order to generate instances for optimal power flow by:

```
def extract_opf_data(mat_file_path): 
    # Convert MATPOWER case to pandapower network 
    net = pc.from_mpc(
        mat_file_path, f_hz=50, casename_mpc_file="mpc", validate_conversion=False
    )
```

where mat_file_path is a path that leads to case9Tree.m from [https://www.dropbox.com/scl/fi/ohc1xd4g03u1ch0356zgj/Tree-Library.zip?dl=0&e=1&file_subpath=%2FTree+Library&rlkey=crey52c766o03chsxgjytsrfa]. Now if I want to access the generator indices by:

```
generator_indices = {bus - 1 for bus in net.gen["bus"].values}
````

I just get two indices {0, 1} instead of {0, 1, 2}. From the pure matlab file, I can see that 3 generators should be returned.

What am I doing wrong here ?

Thank you and best,

Paul

### Label

- [ ] Relevant labels are selected

KS-HTK, 2024-07-01:
Using the [case9.m](https://raw.githubusercontent.com/MATPOWER/matpower/master/data/case9.m) from matpower repo to test. I think what happend here is that the first gen is converted to an ext_grid. This ext_grid is connected to bus 0. But I am not sure if that is desired behavior, it is not Documented.
This also matches with the case9 from pandapower.networks See [Docs](https://pandapower.readthedocs.io/en/latest/networks/power_system_test_cases.html#case-9).

Paulnkk, 2024-07-09:
@KS-HTK thanks for the response but what does that technically mean? Is this bus irrelevant for power generation and not considered in the objective function (as costs). I am not too familiar currently with the technical aspects since I am from the applied mathematics side and try to solve the sdp relaxation from here: https://ieeexplore.ieee.org/document/7056568 (2a) - (2i) with (2i) as rotated second-order-cone

Paulnkk, 2024-07-10:
One question; is the only case that generator buses are not extracted appropriately from the files that they can end-up in `net.ext_grid` or is it possible that other buses which are not generators are also possible to end-up here.

vogt31337, 2024-07-26:
@Paulnkk, I am not sure If I get your problems and questions. So in pandapower you have static generators and generators. You need to decide which one to use.
For solving the loadflow equations you NEED a slack bus aka ext_grid. This table ONLY contains ext_grids, nothing else. There are networks containing multiple slacks.

But I think you want to hear, that you don't need to have an entry in the ext_grid, since you can set a flag on one generator to let it act as a ext_grid.
See https://pandapower.readthedocs.io/en/latest/elements/gen.html, it is in the parameter list but missing in the documentation below...

Paulnkk, 2024-07-29:
@vogt31337 thanks for your response! My problem is that I want to solve the radial opf network from here with a self-written customized solver: https://ieeexplore.ieee.org/document/7056568 specifically (2a) - (2i) with (2i) relaxed as rotated second-order cone. To initialize my instances for the solver, I need the information which buses are referring to generator buses. I will assume that all buses that will be included in the `ext_grid` as slack buses are automatically generators (from what I have seen, this is in most cases true).

vogt31337, 2024-07-29:
@Paulnkk, it's Not incorrect documentation. IT IS how the powerflow works: See Page 35 in https://matpower.org/docs/MATPOWER-manual-8.0.pdf for example.
To be able to define the loadflow equations, there is one "slack Bus" Type == 3 needed. That is Not a Generator, it is the slack Bus.
Regarding your gen stuff I will Take a Look later into your source.
BR

Paulnkk, 2024-07-30:
@vogt31337 Thanks for the response, I understand now

For my use-case, it is important to know which buses are generators and which are not and I encountered the case based on my datasets mentioned above that generator buses were slack buses and therefore ended up in `ext_grid` but not in `net.gen["bus"].values` (which refers to the generator bus indices from what I understand). Here I assumed that this bus would also appear `net.gen["bus"].values` and not only in `ext_grid`

vogt31337, 2024-07-30:
@Paulnkk, no that is not correct. Pandapower is based on matpower (or pypower), so the matrix describing gen and sgens are internally converted to the bus matrix, and the type is set to either type = 1 or type = 2. The slack bus is special since it is needed to solve the equations, otherwise the whole system would be ill conditioned, and therefore it has type = 3 and normally only one exists. Thats why it has its own table. There is also support for distributed slack in pandapower.
Also the slack bus has always pu = 1.0 and va = 0° by definition.

Propably the best for your application would be to look into the ppc / ppci structure. Because there you get this bus table and can simply analyse the type field. This is compatible to the matpower definitions.

closed 2025-11-24 by heckstrahler