e2nIEE/pandapower #2326: Initialise TrafoController fails if transformer is not in_service

bryjohnston, 2024-07-03:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
net = pp.networks.mv_oberrhein()

#switch transformer out of service
net.trafo.loc[114, 'in_service'] = False

trafo_controller = pp.control.controller.trafo_control.TrafoController(net=net, tid=114, side='lv', in_service=True, tol=0.001, trafotype='2W')
```

### Issue Description and Traceback

TrafoController creation fails to initialise if the relevant transformer is not in_service. While this is easily fixable for small scripts, it impacts other functions, especially `from_cim`

KeyError: None

### Expected Behavior

TrafoController should be successfully created. 

### Installed Versions

* python version: 3.11.4 * pandas version:  2.2.2

### Label

- [ ] Relevant labels are selected

pawellytaev, 2024-08-29:
Hi @bryjohnston, I replicated your example and I got the same issue, for TrafoController, DiscreteTapControl and ContinuousTapControl. Just to be sure, is this the error message that you get?

 File "C:\Users\e2n037\git\pandapower_mfisch\pandapower\control\controller\trafo\DiscreteTapControl.py", line 42, in __init__
    super().__init__(net, tid, side, tol=tol, in_service=in_service, level=level, order=order, trafotype=trafotype,
  File "C:\Users\e2n037\git\pandapower_mfisch\pandapower\control\controller\trafo_control.py", line 50, in __init__
    self._set_tap_parameters(net)
  File "C:\Users\e2n037\git\pandapower_mfisch\pandapower\control\controller\trafo_control.py", line 133, in _set_tap_parameters
    self.tap_min = read_from_net(net, self.trafotable, self.controlled_tid, "tap_min", self._read_write_flag)
  File "C:\Users\e2n037\git\pandapower_mfisch\pandapower\auxiliary.py", line 515, in read_from_net
    return _read_from_single_index(net, element, variable, index)
  File "C:\Users\e2n037\git\pandapower_mfisch\pandapower\auxiliary.py", line 584, in _read_from_single_index
    return net[element].at[index, variable]
  File "C:\Users\e2n037\.conda\envs\py310\lib\site-packages\pandas\core\indexing.py", line 2419, in __getitem__
    return super().__getitem__(key)
  File "C:\Users\e2n037\.conda\envs\py310\lib\site-packages\pandas\core\indexing.py", line 2371, in __getitem__
    return self.obj._get_value(*key, takeable=self._takeable)
  File "C:\Users\e2n037\.conda\envs\py310\lib\site-packages\pandas\core\frame.py", line 3877, in _get_value
    row = self.index.get_loc(index)
  File "C:\Users\e2n037\.conda\envs\py310\lib\site-packages\pandas\core\indexes\base.py", line 3654, in get_loc
    raise KeyError(key) from err
KeyError: None

pawellytaev, 2024-08-29:
imho the controller should be added, but set to in_service=False if the trafo is out of service, otherwise this might lead to convergence issues

bryjohnston, 2024-08-29:
Yes, this is the error I had. I did manage to fix it myself for personal use by the same method you've desctribed, but I am not sure my strategy was ideal.

pawellytaev, 2024-08-29:
would you mind opening a PR with your fix? :-) Or share here how exactly you did it?

bryjohnston, 2024-08-29:
My git repository seems to have been corrupted since then so I can't view the differences but I think I made these changes in the trafo_controller.py, I separated the controlled_id from the control flag so there is not a conflict between these two parameters

[image]

pawellytaev, 2024-09-04:
Hi @bryjohnston, I created a PR that should fix this issue, feel free to test it on your code if it works

pawellytaev, 2024-11-25:
closing as respective PR is merged

closed 2024-11-25 by pawellytaev