e2nIEE/pandapower #2329: Asymmetric load into balanced load flow

sicatca, 2024-07-04:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Is there a way to use asymmetric load into balanced load flow? I.e., assigning loads only on phases A or B or C and still run balanced load flow? code works, but the results being publised on res_line has the current still split onto 3 phases.

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-07-26:
I think currently pandapower does not have a three phase calculation. You could emulate this, by designing everything as a three phase system.
Maybe @rbolgaryn do you have an idea?

vogt31337, 2024-11-21:
@sicatca is this still relevant?