e2nIEE/pandapower #2333: [bug] select_subnet loses dataframes

KS-HTK, 2024-07-05:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
from pandapower.toolbox import select_subnet
import pandas as pd

net1 = pp.create_empty_network()
b1 = pp.create_bus(net1, vn_kv=110)
b2 = pp.create_bus(net1, vn_kv=220)

net1["example"] = pd.DataFrame([(1, 2, 3)], columns=["a", "b", "c"])

print(net1)

print(select_subnet(net1, [b1]))
```

### Issue Description and Traceback

Any unknown DataFrame is droped by select_subnet. This is not a good behavior as one would expect that any unknown DataFrame is just copied to the resulting subnet.

Of course this could lead to invalid data being in the subnet but it is better to have data that is no longer valid (for example if it references a bus that is not part of the subnet) than loosing information all together.
The Documentation should mention this behavior regardless of the chosen option.

### Expected Behavior

I would expect the `example` DataFrame to be present in the selected subnet.

### Installed Versions

* python version: 3.9.13
* pandas version: 2.2.2
* networkx version: 3.2.1
* scipy version: 1.13.1
* numpy version: 2.0.0
* packaging version: 24.1
* tqdm version: 4.66.4
* deepdiff version: 7.0.1
* Operating System name/version: Windows 10 Enterprise / 22H2

### Label

- [X] Relevant labels are selected

vogt31337, 2024-07-05:
maybe we should introduce a switch copy_unused_dataframes, which would simply copy such dataframes into all subnets, without touching them.

SteffenMeinecke, 2024-07-08:
There already is the flag `keep_everything_else`. @KS-HTK, you can improve the docstring to avoid that the next user is missing the existing functionality.

PS: please close the issue if it can be closed.

closed 2024-11-19 by KS-HTK