e2nIEE/pandapower #2335: from_json() does not convert net.bus_geodata and net.line_geodata from stored (older) files

SteffenMeinecke, 2024-07-08:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
net = pp.from_json("v2.14.6_net.json")
print(net)
print(net.bus_geodata.head())
print(net.bus.geo.head())
print(net.line.geo.head())
```

### Issue Description and Traceback

convert_format() needs to be extended.

### Expected Behavior

geo data should be converted to the new columns `net.bus.geo` and `net.line.geo`. Instead the data still is in the old dataframes bus_geodata and line_geodata.

### Installed Versions

* python version: * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:

### Label

- [X] Relevant labels are selected

SteffenMeinecke, 2024-08-21:
[case9.json](https://github.com/user-attachments/files/16695006/case9.json)
Attached is a case9 json file with `net.bus_geodata` of a former pp version. Using the following code, I don't see the expected output. @KS-HTK @vogt31337 shouldn't this be fixed by above PR merged?
```
import pandapower as pp
net = pp.from_json(sm.pjoin(sm.desktop, "case9.json"))
if bus_geodata_exists := "bus_geodata" in net.keys():
    print(net.bus_geodata)
    print("net.bus_geodata should not exist anymore.")
if net_bus_geo_is_all_nan := net.bus.geo.isnull().all():
    print("Instead, net.bus.geo is all NaN.")
if not bus_geodata_exists and not net_bus_geo_is_all_nan:
    print("Bus geo data seems to be converted to the current format. Please ensures this also for net.line.geo.")
```

KS-HTK, 2024-11-19:
This seems to be a Bug in the from_json_string function of file_io.py. When it uses json.loads on the file content the resulting network has a format_verison set to `'3.0.0'` resulting in the converter not converting anything.

KS-HTK, 2024-11-19:
@SteffenMeinecke why did you remove the code that does the conversion in [Commit 57c60a1](https://github.com/e2nIEE/pandapower/commit/57c60a1c78e6345ffa0346c8c4c68081c0a0f9c5#diff-804c47c1650b3c47526d3f906f9d3b713842a57f8e88be7ae77486ccb9b4ec98L49)

This was fixed until that commit removed the code that does the conversion.

SteffenMeinecke, 2024-11-19:
Dear @KS-HTK,
I cannot remember. Thanks for finding out the reason for failure. I just raised the PR https://github.com/e2nIEE/pandapower/pull/2452, please review.

commit 02984641 (2024-11-19): fixes #2335
closed 2024-11-19 by SteffenMeinecke