e2nIEE/pandapower #2337: [bug] reference_column assigned incorrectly on creation of groups

awarsewa, 2024-07-10:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp

net = pp.networks.example_simple()
pp.create_group(net, ['switch', 'line'], [[0, 1], [0, 1]])
pp.attach_to_group(net, 0, ['switch'], [[2]])
```

### Issue Description and Traceback

When you create a group, the "reference_column" of the newly created group is set to NaN.
On adding an element, comparison of NaN to None in attach_to_group yields different reference columns.
This leads to a "nan"-column being added to net.switch in groups.py line 975:
```python
net[et][reference_column].loc[idxs] = ["%s_%i_%s" % (et, idx, str(
                    uuid.uuid4())) for idx in idxs]
```

### Expected Behavior

Check for different reference columns in attach_to_group should either handle NaN or entries in reference_column of a newly created group should be set to None instead.

### Installed Versions

INSTALLED VERSIONS
------------------
commit                : d9cdd2ee5a58015ef6f4d15c7226110c9aab8140
python                : 3.12.4.final.0
python-bits           : 64
OS                    : Windows
OS-release            : 10
Version               : 10.0.19045
machine               : AMD64
processor             : Intel64 Family 6 Model 140 Stepping 1, GenuineIntel
byteorder             : little
LC_ALL                : None
LANG                  : en
LOCALE                : de_DE.cp1252

pandas                : 2.2.2
numpy                 : 1.26.4
pytz                  : 2024.1
dateutil              : 2.9.0.post0
setuptools            : 69.5.1
pip                   : 24.0
Cython                : None
pytest                : None
hypothesis            : None
sphinx                : None
blosc                 : None
feather               : None
xlsxwriter            : None
lxml.etree            : None
html5lib              : None
pymysql               : None
psycopg2              : None
jinja2                : None
IPython               : 8.26.0
pandas_datareader     : None
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
gcsfs                 : None
matplotlib            : 3.9.0
numba                 : 0.60.0
numexpr               : None
odfpy                 : None
openpyxl              : 3.1.5
pandas_gbq            : None
pyarrow               : None
pyreadstat            : None
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : None
xarray                : None
xlrd                  : None
zstandard             : None
tzdata                : 2024.1
qtpy                  : None
pyqt5                 : None

### Label

- [X] Relevant labels are selected

vogt31337, 2024-11-21:
@awarsewa I do not really understand your problem. I think you have to create a group first. But I will do maintenance on the grouping features in pandapower shortly.

awarsewa, 2024-11-21:
Thanks for looking into it. As you can see in the example code above, I have created a group in the first step and then tried adding an element to it. This is when I experienced some odd behaviour with a NaN column being added to the net.switch table. I might have misunderstood how you're supposed to use the grouping feature, but it looks like a bug to me.

closed 2026-11-11 by heckstrahler