e2nIEE/pandapower #2338: Convergence for power flow analysis shows inconsistent properties

Paulnkk, 2024-07-11:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hey,

I am currently running power flow analysis in order to generate the admittance matrix Ybus:

```
# Convert MATPOWER case to pandapower network
# mat_file_path is file to tree files
net = pc.from_mpc(
        mat_file_path, f_hz=50, casename_mpc_file="mpc", validate_conversion=False
    )
# Run power flow to calculate the Ybus matrix
pp.runpp(net)
Ybus = net._ppc["internal"]["Ybus"]
# Extract the conductance (G) and susceptance (B) matrices
G = Ybus.real
B = Ybus.imag
```
Now I am facing the issue, when I run those lines for the "case14Tree.m" that I receive the following error message:

```
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/create.py:5049: FutureWarning: The behavior of DataFrame concatenation with empty or all-NA entries is deprecated. In a future version, this will no longer exclude empty or all-NA columns when determining the result dtypes. To retain the old behavior, exclude the relevant entries before the concat operation.
  net[table] = pd.concat([net[table], dd[dd.columns[~dd.isnull().all()]]], sort=False)
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/converter/pypower/from_ppc.py:199: RuntimeWarning: divide by zero encountered in divide
  max_i_ka = ppc['branch'][:, 5]/ppc['bus'][to_bus, BASE_KV]/np.sqrt(3)
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/converter/pypower/from_ppc.py:210: RuntimeWarning: divide by zero encountered in divide
  c_nf_per_km=(ppc['branch'][is_line, BR_B]/Zni[is_line]/omega*1e9/2).real,
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/converter/pypower/from_ppc.py:210: RuntimeWarning: invalid value encountered in divide
  c_nf_per_km=(ppc['branch'][is_line, BR_B]/Zni[is_line]/omega*1e9/2).real,
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/converter/pypower/from_ppc.py:277: FutureWarning: Setting an item of incompatible dtype is deprecated and will raise an error in a future version of pandas. Value '[]' has dtype incompatible with int64, please explicitly cast to a compatible dtype first.
  branch_lookup.loc[~is_line, "element"] = idx_trafo
numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/build_branch.py:164: RuntimeWarning: invalid value encountered in divide
  branch[f:t, BR_R] = line["r_ohm_per_km"].values * length_km / baseR / parallel
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/build_branch.py:165: RuntimeWarning: invalid value encountered in divide
  branch[f:t, BR_X] = line["x_ohm_per_km"].values * length_km / baseR / parallel
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/build_branch.py:194: RuntimeWarning: invalid value encountered in multiply
  b = 2 * net.f_hz * math.pi * line["c_nf_per_km"].values * 1e-9 * baseR * length_km * parallel
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/build_branch.py:206: RuntimeWarning: invalid value encountered in multiply
  branch[f:t, RATE_A] = max_load / 100. * max_i_ka * df * parallel * vr
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/build_bus.py:487: RuntimeWarning: invalid value encountered in divide
  v_ratio = (ppc["bus"][bus_lookup[s["bus"].values], BASE_KV] / s["vn_kv"].values) ** 2 * base_multiplier
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/pypower/makeBdc.py:97: RuntimeWarning: invalid value encountered in divide
  b = stat / branch[:, BR_X]  ## series susceptance
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/scipy/sparse/linalg/_dsolve/linsolve.py:293: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/pypower/makeYbus.py:82: RuntimeWarning: invalid value encountered in divide
  Ysf = stat / (branch[:, BR_R] + 1j * branch[:, BR_X])  ## series admittance
/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/pypower/dSbus_dV.py:34: RuntimeWarning: invalid value encountered in divide
  diagVnorm = sparse((V / abs(V), (ib, ib)))
Traceback (most recent call last):
  File "/Users/PaulKandora/PyCharm_projects/CopOpf/copopf/test_soc_on_large_instances.py", line 124, in <module>
    opf_data = extract_opf_data('Tree_Library/case14Tree.m')
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/PaulKandora/PyCharm_projects/CopOpf/copopf/extract_opf_data_from_matlab.py", line 13, in extract_opf_data
    pp.runpp(net)
  File "/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/run.py", line 250, in runpp
    _powerflow(net, **kwargs)
  File "/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/powerflow.py", line 86, in _powerflow
    _ppci_to_net(result, net)
  File "/Users/PaulKandora/opt/anaconda3/envs/copopf/lib/python3.12/site-packages/pandapower/powerflow.py", line 190, in _ppci_to_net
    raise LoadflowNotConverged("Power Flow {0} did not converge after "
pandapower.powerflow.LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!
```
I assumed that the issue was related to dimensionality of the dataset but when I run the code on higher dimensional cases such as "case30Tree.m", I receive the correct results. Probably, it is not the best approach to generate admittance matrix (grateful if you know a better way to do so) but not sure how to resolve this issue.

Thanks in advance and best regards,

Paul

### Label

- [ ] Relevant labels are selected

SteffenMeinecke, 2024-07-12:
Dear @Paulnkk, 
I agree that numerical issues might arise when trying to solve the power flow with the converted grid data. You can try to run `pp.diagnostic(net)` after receiving the data from matpower conversion. Maybe some r, x, or b values are extreme, for some reason.

Paulnkk, 2024-07-14:
@SteffenMeinecke thanks a lot for the suggestion, I will try the approach you proposed and try to fix the issue

vogt31337, 2024-11-21:
@Paulnkk still relevant? I have seen in another issue, that pp does something with zero impedance lines, which seem to be part of the case14 maybe this is a hint why your results are weird.

Paulnkk, 2024-11-21:
@vogt31337 thanks for the response! Yes this case is still relevant for me, I planned to check in detail tomorrow

Paulnkk, 2024-11-22:
@vogt31337 I checked and the issue mainly comes from the fact that `baseKV` in my ieee 14 bus case is 0. I am considering radial networks from here: https://www.dropbox.com/scl/fi/ohc1xd4g03u1ch0356zgj/Tree-Library.zip?rlkey=crey52c766o03chsxgjytsrfa&e=1&dl=0

I will try to use the `baseKV` from the standard MATPOWER case file for the IEEE 14-bus system and check if results are correcr