e2nIEE/pandapower #2340: test failure: 'csc_matrix' object has no attribute

quant12345, 2024-07-12:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

When testing I get multiple errors, for example:
```
FAILED pandapower/test/converter/test_from_ppc.py::test_to_and_from_ppc - AttributeError: 'csc_matrix' object has no attribute 'H'
FAILED pandapower/test/loadflow/test_runpp.py::test_bsfw_algorithm_with_branch_loops - AttributeError: 'csc_matrix' object has no attribute 'A'
```

### Label

- [ ] Relevant labels are selected

quant12345, 2024-07-13:
Figured it out because of the new scipy. In the new .A and
.H attributes were removed. Works since scipy 1.13.0.

vogt31337, 2024-07-26:
@quant12345 good to know, currently we fixed the scipy version. Do you know if there is a fix to get these matrices again?

quant12345, 2024-07-26:
> @quant12345 good to know, currently we fixed the scipy version. Do you know if there is a fix to get these matrices again?

@vogt31337 I haven't done this yet, but I plan to try it in a while.

PR #2358 (2024-08-06): Replacing deprecated H and A from scipy.sparse.csc_matrix
Fixes: #2340

Used csc_matrix documentation [csc_matrix.A](https://docs.scipy.org/doc/scipy-1.13.0/reference/generated/scipy.sparse.csc_matrix.A.html#scipy.sparse.csc_matrix.A) and [csc_matrix.H](https://docs.scipy.org/doc/scipy-1.13.0/reference/generated/scipy.sparse.csc_matrix.H.html#scipy.sparse.csc_matrix.H)

Quote from release SciPy 1.14.0: "Additionally, the .A and.H attributes were [removed."](https://github.com/scipy/scipy/releases/tag/v1.14.0)

Tested on `scipy version  1.14.0`

tests to check:
```
pytest pandapower/test/conver [... cut]
closed 2024-08-06 by vogt31337