e2nIEE/pandapower #2345: [bug] With new version of numpy, issue with library specifically imports

bimbimbambam69, 2024-07-15:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
import pandapower.plotting as plot
import matplotlib.pyplot as plt

def oc_relay_net():
       # create an empty network
    net = pp.create_empty_network()

    # create buses
    pp.create_buses(net, nr_buses=7, vn_kv=20, index=[0, 1, 2, 3, 4, 5, 6], name=None, type="n",
                    geodata=[(0, 0), (0, -1), (-2, -2), (-2, -4), (2, -2), (2, -3), (2, -4)])

    # create external grids
    pp.create_ext_grid(net, 0, vm_pu=1.0, va_degree=0, s_sc_max_mva=100, s_sc_min_mva=50, rx_max=0.1, rx_min=0.1)

    pp.create_lines(net, from_buses=[0, 1, 2, 1, 4, 5], to_buses=[1, 2, 3, 4, 5, 6], length_km=[2, 5, 4, 4, 0.5, 0.5],
                    std_type="NAYY 4x50 SE",
                    name=None, index=[0, 1, 2, 3, 4, 5], df=1., parallel=1)

    net.line["endtemp_degree"] = 250

    # Define switches
    pp.create_switches(net, buses=[0, 1, 1, 2, 4, 5], elements=
    [0, 1, 3, 2, 4, 5], et='l', type="CB_DTOC")
    # define load
    pp.create_loads(net, buses=[3, 6], p_mw=[5, 2], q_mvar=[1, 1], const_z_percent=0, const_i_percent=0, sn_mva=None,
                    name=None, scaling=1., index=[0, 1])

    return net

net = oc_relay_net()
from pandapower.protection.protection_devices.ocrelay import OCRelay

for k in range(6):
    OCRelay(net, switch_index=k, oc_relay_type='DTOC', time_settings=[0.07, 0.5, 0.3])

from pandapower.protection.utility_functions import create_sc_bus
import pandapower.shortcircuit as sc
from pandapower.protection.run_protection import calculate_protection_times

# calculate the short circuit
net_sc = create_sc_bus(net, sc_line_id=4, sc_fraction=0.5)
sc.calc_sc(net_sc, bus=max(net_sc.bus.index), branch_results=True)

# calculate the protection times
protection_results = calculate_protection_times(net_sc, scenario='sc')

# show protection_results
print(protection_results)

net.protection.object.at[2].plot_protection_characteristic(net=net, xmin=10, xmax=10000, ymin=0.01, ymax=10000)
plt.grid(True)
plt.xlabel("Ток [А]")
plt.ylabel("Время [с]")
plt.title("Relay")
plt.show()
```

### Issue Description and Traceback

The issue with new version of numpy (2.0.0) is importing Inf, while new numpy need to import inf started with small letter

Traceback (most recent call last):
  File "E:\PyCharmProjects\pythonProject\````````.py", line 1, in <module>
    import pandapower as pp
  File "C:\Users\Lolik\AppData\Local\Programs\Python\venv\Lib\site-packages\pandapower\__init__.py", line 11, in <module>
    from pandapower.powerflow import *
  File "C:\Users\Lolik\AppData\Local\Programs\Python\venv\Lib\site-packages\pandapower\powerflow.py", line 13, in <module>
    from pandapower.pf.run_bfswpf import _run_bfswpf
  File "C:\Users\Lolik\AppData\Local\Programs\Python\venv\Lib\site-packages\pandapower\pf\run_bfswpf.py", line 20, in <module>
    from pandapower.pypower.newtonpf import _evaluate_Fx, _check_for_convergence
  File "C:\Users\Lolik\AppData\Local\Programs\Python\venv\Lib\site-packages\pandapower\pypower\newtonpf.py", line 14, in <module>
    from numpy import float64, array, angle, sqrt, square, exp, linalg, conj, r_, Inf, arange, zeros, \
ImportError: cannot import name 'Inf' from 'numpy' (C:\Users\Lolik\AppData\Local\Programs\Python\venv\Lib\site-packages\numpy\__init__.py). Did you mean: 'inf'?

### Expected Behavior

Can`t run any code with this traceback after attempt to impor pandapower

### Installed Versions

* python version: 3.12.4.final.0 * pandas version: 2.2.2  * scipy version: 1.13.0 * numpy version: 2.0.0 * Operating System name/version: Windows/10.0.19045

### Label

- [X] Relevant labels are selected

bimbimbambam69, 2024-07-15:
just need to rewrite imports from numpy in pandapower library from Inf to inf

bimbimbambam69, 2024-07-16:
Official changelog of numpy 2.0.0:
**Alias np.Inf has been removed. Use np.inf instead.**

pawellytaev, 2024-08-29:
Hi, this will be fixed in the upcoming release, see the current changelog on pandapower develop: 
**[FIXED] namespace changes from numpy 2.0 release**

vogt31337, 2024-11-21:
should be fixed in latest pp release 2.14.11 and will be updated in 3.0.0

closed 2024-11-21 by vogt31337