e2nIEE/pandapower #2350: Parameter "parallel" in create_line

ricoeo, 2024-07-21:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

For the pandapower.create_line, it accepts parameter parallel, I am not sure I get the parameter correctly. 
For example, I get two parallel lines between bus 0, and bus 1. then 
is it pp.create_line_from_parameters(net, from_bus=0, to_bus=1,length_km=line_data.at[idx, "length_km"],r_ohm_per_km=line_data.at[idx, "r_ohm_per_km"],..., **parallel=2**) the same effect as 
I create **two lines with both** pp.create_line_from_parameters(net, from_bus=0, to_bus=1,length_km=line_data.at[idx, "length_km"],r_ohm_per_km=line_data.at[idx, "r_ohm_per_km"],..., parallel=1). 

I notice the problem of the parallel system while I try to convert it into the networkx graph, this line has the property "path" equals 1 (multi sets to True while creating the graph). Can I get an answer?

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-07-26:
I don't understand your question. This parallel feature doesn't create a new line in tandem to the other one. It simply sets a value, which the loadflow then uses to change the line paramter during a calculation.

ricoeo, 2024-07-26:
> I don't understand your question. This parallel feature doesn't create a new line in tandem to the other one. It simply sets a value, which the loadflow then uses to change the line paramter during a calculation.

Thanks for your answer. I don’t quite get the meaning „use to change the line parameter during a calculation“. Is it possible you can explain it. And the correct way to introduce parallel lines between two buses is to create two lines from these two buses. for example ,create_ line(from bus=0, to bus=1…, parallel=1) and do it  again create_ line(from bus=0, to bus=1…, parallel=1). Do I get it correct?

pawellytaev, 2024-09-05:
Hi @ricoeo, in pandapower there are 2 ways of creating parallel lines, either with your method or by the parameter `parallel`. 
It is considered in the calculation of the `ppc` branch tables, here are a few examples how it is used:
https://github.com/e2nIEE/pandapower/blob/11af12cd3db5a85135c7c022e8f5ac2e7a5b4086/pandapower/build_branch.py#L208
https://github.com/e2nIEE/pandapower/blob/11af12cd3db5a85135c7c022e8f5ac2e7a5b4086/pandapower/build_branch.py#L238

In both cases the node admittance matrix will be basically the same, hence the results will be the same. You will just see in `res_line` with your method, two lines with the same results, with `parallel=2`, one line with the exact same values. Feel free to try this out. 

Basically, which method you use is up to you and your preference.

closed 2024-09-19 by ricoeo