e2nIEE/pandapower #2352: Merge two networks as DSO/TSO network?

samabu2011, 2024-07-22:
Hi

I was wondering if and how you can use the merge function to merge two networks such that one functions as a dso and one as TSO?

vogt31337, 2024-07-26:
@samabu2011 you can merge two networks. It should work, but you have to identify the merging points / busses. What exactly do you mean with one functions as dso and one as tso?

samabu2011, 2024-07-26:
Thanks for the answer! I would like to construct an example where I can use two of the IEEE cases from here: https://pandapower.readthedocs.io/en/v2.1.0/networks/power_system_test_cases.html#pandapower.networks.case24_ieee_rts and build a new network with them. Ideally one would represent a TSO and one would respresent DSO, such that I can do some further simulations. Currently for example I am trying to connect case14() as a TSO with case4gs() playing the role as a DSO. 

How owould you identify the merging points and busses?

vogt31337, 2024-11-21:
> How would you identify the merging points and busses?

@samabu2011 I cannot tell you that, as far as I know, at such points 3 winding transformer are used. HV Side is your TSO Grid, MV Side is your DSO Grid, and LV Side is for suppyling the substation.