e2nIEE/pandapower #2381: Difficulty adding storage to DG based inverter

HafizRiaz, 2024-08-27:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Dear Author, i am working on Distribution system voltage control with DRL (machine learning algo) using Inverter based PV and SVC, which gives the expected results. however, when i try to replace SVC with Storage system , it gives error. I know that way , storage system is defined in pndapower is quite different from SVC which is created the same command of "pp.create_sgen" for Inverter based PV. The error is   as uner. Plz help me resolving the issue . Although dimension are same

Traceback (most recent call last):
  File "G:\Mehboob\Single-Period\DDPG_OSTC_store.py", line 694, in <module>
    agent.train(num_frames, plotting_interval=400000)
  File "G:\Mehboob\Single-Period\DDPG_OSTC_store.py", line 400, in train
    next_state, reward, done, violation, violation_M, violation_N,voltage_M, voltage_N, grid_loss, new_state = self.step_model(action)
  File "G:\Mehboob\Single-Period\DDPG_OSTC_store.py", line 275, in step_model
    next_state, reward, done, violation, violation_M, violation_N, voltage_M, voltage_N, grid_loss, new_state = self.env.step_model(action)
  File "G:\Mehboob\Single-Period\Env2.py", line 218, in step_model
    self.model.sgen.q_mvar = action #  clipped action values are assigned to Q power attribute of sgen
  File "C:\Users\sa\anaconda3\envs\myenv\lib\site-packages\pandas\core\generic.py", line 5931, in __setattr__
    self[name] = value
  File "C:\Users\sa\anaconda3\envs\myenv\lib\site-packages\pandas\core\frame.py", line 3980, in __setitem__
    self._set_item(key, value)
  File "C:\Users\sa\anaconda3\envs\myenv\lib\site-packages\pandas\core\frame.py", line 4174, in _set_item
    value = self._sanitize_column(value)
  File "C:\Users\sa\anaconda3\envs\myenv\lib\site-packages\pandas\core\frame.py", line 4915, in _sanitize_column
    com.require_length_match(value, self.index)
  File "C:\Users\sa\anaconda3\envs\myenv\lib\site-packages\pandas\core\common.py", line 571, in require_length_match
    raise ValueError(
ValueError: Length of values (4) does not match length of index (3)

### Label

- [X] Relevant labels are selected

pawellytaev, 2024-08-29:
Hi, can you provide a minimal example code so that we can replicate the issue?
To me, this looks more like an issue in assigning the agent's actions (`len(action)=4`) to the sgens (`len(net.sgen)=3`) and less like a specific pandapower issue

HafizRiaz, 2024-08-30:
Thank you for the response. here is the example code. Actually the code goes well when i add SVC to Inveter based PV static generator. However when i try to replace SVC with storage it gives error where i am adding its reactive power so that it may be assigned to action values of DRL agent for volt/var control. I think nature of storage is different from that of SVC. In literature where storage is modeled, just active power and SOC are mentioned but in pandapower modeling , storage creation also has the provision of defining minimum and maximum reactive power which can be used in volt-var control strategy, If you could suggest me some related help points, i will highly appreciate. Stay blessed and healthy. 
[image]
this section (image attached) where we want to clip actions causes error 
[image]

pawellytaev, 2024-09-04:
It still looks to me that your dimenstion of action is different from the number of sgens. In your `__init__` I see that you set the action dimenstion to be `len(self.id_iber) + len(self.id_store)`. Then the DRL agent calculates the actions (I assume both for storages and sgens). Then you try to assign this action to only sgens, which throws the error. I propose you work with masked arrays.

Since this is not a specific pandapower issue to me, I'll close this issue, but you can still comment here

closed 2024-09-04 by pawellytaev