e2nIEE/pandapower #2382: [bug] Using minimal_example code results in incorrect result information PandaPower Version 2.14.6

karatekid2501, 2024-08-27:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

I am testing PandaPower and for some reason when inputting the minimal example code into my project and test the results of the buses I'm getting these results:

vm_pu | va_degree | p_mw | q_mvar
-- | -- | -- | --
1.020000 | 0.000000 | -0.107265 | -0.052675
1.008843 | -150.760126 | 0.000000 | 0.000000
0.964431 | -149.884141 | 0.100000 | 0.050000

Instead of this result

vm_pu | va_degree | p_mw | q_mvar
-- | -- | -- | --
1.020000 | 0.000000 | -0.107265 | -0.052675
1.008843 | -0.760126 | 0.000000 | 0.000000
0.964431 | 0.115859 | 0.100000 | 0.050000

As you can see for some reason the va_degree in both bus 2 and 3 are not the same. I've doubled check on the code to make sure that the code is correct and it is:

```
import pandapower as pp
def Example():
    try:
        net = pp.create_empty_network()
    
        #create buses
        bus1 = pp.create_bus(net, vn_kv=20., name="Bus 1")
        bus2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")
        bus3 = pp.create_bus(net, vn_kv=0.4, name="Bus 3")

        #create bus elements
        pp.create_ext_grid(net, bus=bus1, vm_pu=1.02, name="Grid Connection")
        pp.create_load(net, bus=bus3, p_mw=0.100, q_mvar=0.05, name="Load")

        #create branch elements
        trafo = pp.create_transformer(net, hv_bus=bus1, lv_bus=bus2, std_type="0.4 MVA 20/0.4 kV", name="Trafo")
        line = pp.create_line(net, from_bus=bus2, to_bus=bus3, length_km=0.1, std_type="NAYY 4x50 SE", name="Line")
        
        #return currentNetwork.bus.to_string()
        pp.runpp(net)
        return net.res_bus

        
    except Exception:
        return "Error Thrown"
```

So I am completely stuck on what the issue could be. Any help is appreciated.

### Label

- [X] Relevant labels are selected

pawellytaev, 2024-08-30:
Hi @karatekid2501, could you share your net.trafo.shift_degree? You see there is an exact 150 degree voltage angle difference between your results and the results from the tutorial. if you set your trafo shift to 0, there should be now difference.

Thanks for bringing that up, it's a bug in the minimal example tutorial :-)

karatekid2501, 2024-08-30:
> Hi @karatekid2501, could you share your net.trafo.shift_degree? You see there is an exact 150 degree voltage angle difference between your results and the results from the tutorial. if you set your trafo shift to 0, there should be no difference.
> 
> Thanks for bringing that up, it's a bug in the minimal example tutorial :-)

Hi,

Thanks for getting back to me, at least this confirms that I am not going mad and it is a bug. 

When printing net.trafo.shift_degree, it is reporting back 150. I'm assuming I would need to use the create_transformer_from_parameters function but I am unsure what to put in the extra parameters required to ensure that the results match the minimal tutorial outcome.

pawellytaev, 2024-09-02:
Hi @karatekid2501, this should work: 
```
traf0 = pp.create_transformer_from_parameters(net, hv_bus=bus1, lv_bus=bus2, name="Trafo", sn_mva=0.4, vn_hv_kv=20.0, 
                                              vn_lv_kv=0.4, vk_percent=6.0, vkr_percent=1.425, pfe_kw=1.35, i0_percent=0.3375,
                                              shift_degree=0, tap_min=-2, tap_max=2, tap_pos=0)
```
I basically copied all the trafo parameters from the net.trafo table but made `shift_degree=0`. You can find more information on how to create customized elements here: https://github.com/e2nIEE/pandapower/blob/develop/tutorials/create_advanced.ipynb

closed 2024-09-04 by pawellytaev