e2nIEE/pandapower #2392: [bug] IEEE14 Case: Zero resistance lines get interpreted as transformers

leuchtum, 2024-09-05:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower
import pandas as pd
import pypowsybl

pandapower_net = pandapower.networks.case14()
pypowsybl_net = pypowsybl.network.create_ieee14()

results = {
    "n_buses": {
        "expected": 14,
        "powsybl": len(pypowsybl_net.get_buses()),
        "pandapower": len(pandapower_net.bus),
    },
    "n_lines": {
        "expected": 17,
        "powsybl": len(pypowsybl_net.get_lines()),
        "pandapower": len(pandapower_net.line),
    },
    "n_trafos": {
        "expected": 3,
        "powsybl": len(pypowsybl_net.get_2_windings_transformers()),
        "pandapower": len(pandapower_net.trafo),
    },
    "n_gens": {
        "expected": 5,
        "powsybl": len(pypowsybl_net.get_generators()),
        "pandapower": len(pandapower_net.gen),
    },
    "n_loads": {
        "expected": 11,
        "powsybl": len(pypowsybl_net.get_loads()),
        "pandapower": len(pandapower_net.load),
    },
    "n_shunts": {
        "expected": 1,
        "powsybl": len(pypowsybl_net.get_shunt_compensators()),
        "pandapower": len(pandapower_net.shunt),
    },
}

print(pd.DataFrame(results).T)

#           expected  powsybl  pandapower
# n_buses         14       14          14
# n_lines         17       17          15
# n_trafos         3        3           5
# n_gens           5        5           4
# n_loads         11       11          11
# n_shunts         1        1           1
```

### Issue Description and Traceback

The IEEE14 case has 3 trafos and 17 lines, resulting in 20 branches (when interpreting it in MATPOWER style). If we look at the example above , pandapower has some trouble.

First, there is one generator less than expected. This is ok, as the missing generator is interpreted as the external grid.

But then if you compare `n_lines`and `n_trafos`, something is odd. Two lines are interpreted as trafos. I had a deeper look into it and I think the problem lies in how a zero-resistance line is interpreted by pandapower when loading a case file. If we have a look at the normal ieee14 data (see [here](https://github.com/caseformat/ieee14/blob/21f2cd1a1f18b046a079d0d54ca2155773d49d05/branch.csv#L15-L16)) there are two lines that have a zero resistance.

### Expected Behavior

Expected result see example above.

### Installed Versions

```
INSTALLED VERSIONS
------------------
commit                : d9cdd2ee5a58015ef6f4d15c7226110c9aab8140
python                : 3.12.5.final.0
python-bits           : 64
OS                    : Darwin
OS-release            : 23.6.0
Version               : Darwin Kernel Version 23.6.0: Mon Jul 29 21:14:21 PDT 2024; root:xnu-10063.141.2~1/RELEASE_ARM64_T8103
machine               : arm64
processor             : arm
byteorder             : little
LC_ALL                : None
LANG                  : en_US.UTF-8
LOCALE                : en_US.UTF-8

pandas                : 2.2.2
numpy                 : 1.26.4
pytz                  : 2024.1
dateutil              : 2.9.0.post0
setuptools            : None
pip                   : 24.2
Cython                : None
pytest                : 8.3.2
hypothesis            : None
sphinx                : None
blosc                 : None
feather               : None
xlsxwriter            : None
lxml.etree            : None
html5lib              : None
pymysql               : None
psycopg2              : None
jinja2                : None
IPython               : None
pandas_datareader     : None
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
gcsfs                 : None
matplotlib            : 3.9.2
numba                 : None
numexpr               : None
odfpy                 : None
openpyxl              : None
pandas_gbq            : None
pyarrow               : None
pyreadstat            : None
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : None
xarray                : None
xlrd                  : None
zstandard             : None
tzdata                : 2024.1
qtpy                  : None
pyqt5                 : None
```

```terminal
> pip freeze
annotated-types==0.7.0
bracex==2.5
bump-my-version==0.26.0
click==8.1.7
contourpy==1.3.0
cycler==0.12.1
deepdiff==8.0.1
fonttools==4.53.1
iniconfig==2.0.0
kiwisolver==1.4.5
-e git+ssh://git@github.com/leuchtum/ma-core.git@e480e47f57173f3d45d811e2edcd0e8721171d17#egg=macore
markdown-it-py==3.0.0
matplotlib==3.9.2
mdurl==0.1.2
mypy==1.11.2
mypy-extensions==1.0.0
networkx==3.3
numpy==1.26.4
orderly-set==5.2.2
packaging==24.1
pandapower==2.14.11
pandas==2.2.2
pandas-stubs==2.2.2.240807
pillow==10.4.0
pimpmyplt==0.0.6
pluggy==1.5.0
prettytable==3.11.0
prompt-toolkit==3.0.36
pydantic==2.8.2
pydantic-settings==2.4.0
pydantic_core==2.20.1
Pygments==2.18.0
pyparsing==3.1.4
pypowsybl @ file:///Users/daniel/Desktop/DEV_ma_core/pypowsybl-1.7.0.dev1-cp312-cp312-macosx_11_0_arm64.whl
pytest==8.3.2
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
pytz==2024.1
questionary==2.0.1
rich==13.8.0
rich-click==1.8.3
ruff==0.6.3
scipy==1.13.1
six==1.16.0
tomlkit==0.13.2
tqdm==4.66.5
types-pytz==2024.1.0.20240417
typing_extensions==4.12.2
tzdata==2024.1
wcmatch==9.0
wcwidth==0.2.13
```

### Label

- [ ] Relevant labels are selected

leuchtum, 2024-09-05:
Supplement:

```python
pandapower.runpp(pandapower_net)
assert pandapower_net["_ppc"]["branch"].shape[0] == 20
```

pawellytaev, 2024-09-05:
Hi @leuchtum, thanks for bringing that up. We made some recent changes in pandapower develop that affected the networks in pandapower, see the PR #2347. Could you test this with the current version of develop and see what happens? Maybe this is already solved there

leuchtum, 2024-09-05:
@pawellytaev Checked on 2.8.0, 2.14.11 and branch develop (commit 11af12cd), all with same result.

pawellytaev, 2024-09-05:
alright, let me take a look into it. I'll come back to this in the next few days

vogt31337, 2025-11-26:
Just had a quick look into the network, I think the case14 is broken or corrupted, since generators and other things are missing.

vogt31337, 2025-12-24:
@leuchtum , I think I found at least one issue: [Alroom Power System Case 14](https://al-roomi.org/power-flow/14-bus-system) describes the transformer between bus 4, 9 and 8 as a three winding transformer. But since this pypower can only model branches between two buses, this 3W transformer get's split up into three branches.